john30/ebusd #204: no signal after update to 3.2

semenoh, 2018-10-04:
Using 3.1 I've been able to read data from the bus. But after update to 3.2 it worked for some time
```
pi@raspberrypi:~ $ ebusctl i
version: ebusd 3.2.v3.2
update check: revision v3.2-8-g0f77a9d available, broadcast.csv: newer version available
signal: acquired
symbol rate: 16
max symbol rate: 83
min arbitration micros: 3218
max arbitration micros: 3327
min symbol latency: 5
max symbol latency: 5
reconnects: 0
masters: 2
messages: 210
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0316;HW=0903", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
buth then stopped to work
```
pi@raspberrypi:~ $ ebusctl i
version: ebusd 3.2.v3.2
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
I suspect that it was using old config files but then it was not able to fetch new configs (as ebusd.eu/config/ is not available) and stopped working

semenoh, 2018-10-04:
I've changed `/etc/default/ebusd` to `EBUSD_OPTS="--configpath=/etc/ebusd/ --scanconfig=full -d /dev/ttyUSB0 -p 8888 --httpport=8080 --loglevel=info"` and it works now as before
```
pi@raspberrypi:~ $ ebusctl i
version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 14
max symbol rate: 52
min arbitration micros: 2090
max arbitration micros: 3343
min symbol latency: 4
max symbol latency: 5
reconnects: 0
masters: 2
messages: 210
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0316;HW=0903", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

So basically issue can be changed to `http://ebusd.eu/config/` is down.

WWebber, 2018-10-05:
Which files you use in /etc/ebusd/ ?
I tried the V2.x.x and V2.1.x files, but both report errors!
ebusd.eu apararently is empty - so how else can I make this working?

john30, 2018-10-05:
no, ebusd.eu is not empty, it works as desired but will not serve to normal browsers.
the missing signal does not have anything to do with the config service.
you might want to check your device ttyUSB0 in the system log as well as the ebusd.log

WWebber, 2018-10-05:
well this is what I see all the time.. 

`2018-10-05 16:50:10.035 [main notice] ebusd 3.2.v3.2 started with auto scan
2018-10-05 16:50:10.574 [bus notice] bus started with own address 31/36
2018-10-05 16:50:10.574 [bus notice] signal acquired
2018-10-05 16:50:10.574 [bus notice] new master 10, master count 2
2018-10-05 16:50:10.575 [bus notice] new master 03, master count 3
2018-10-05 16:50:10.575 [update notice] received unknown MS cmd: 1008b5110101 / 092625a013ff280000ff
2018-10-05 16:50:10.575 [bus notice] new master 71, master count 4
2018-10-05 16:50:10.575 [update notice] received unknown MS cmd: 1076b5110101 / 09ff25a013ff280000ff
2018-10-05 16:50:10.576 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2018-10-05 16:50:10.576 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2018-10-05 16:50:10.577 [update notice] received unknown MS cmd: 1008b5110101 / 092625a013ff280000ff
2018-10-05 16:50:10.578 [update notice] received unknown MS cmd: 1076b5110101 / 09ff25a013ff280000ff
2018-10-05 16:50:10.578 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2018-10-05 16:50:10.578 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2018-10-05 16:50:10.580 [update notice] received unknown MS cmd: 1008b5110101 / 092625a013ff280000ff
2018-10-05 16:50:10.580 [update notice] received unknown MS cmd: 1076b5110101 / 09ff25a013ff280000ff
2018-10-05 16:50:10.581 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2018-10-05 16:50:10.581 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2018-10-05 16:50:10.582 [update notice] received unknown MS cmd: 1008b5110101 / 092625a013ff280000ff
2018-10-05 16:50:10.583 [update notice] received unknown MS cmd: 1076b5110101 / 09ff25a013ff280000ff
2018-10-05 16:50:10.583 [update notice] received unknown BC cmd: 10feb505025c00
2018-10-05 16:50:10.583 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2018-10-05 16:50:10.584 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2018-10-05 16:50:10.584 [update notice] received unknown MS cmd: 1008b5040100 / 0a0306501605100518a013
2018-10-05 16:50:13.554 [update notice] received unknown BC cmd: 10feb5160301a013
2018-10-05 16:50:24.907 [main error] scan config 08: ERR: wrong symbol received
2018-10-05 16:50:34.340 [main error] scan config 15: ERR: arbitration lost
2018-10-05 16:50:36.614 [main error] scan config 76: ERR: wrong symbol received
2018-10-05 16:50:52.299 [main error] scan config 08: ERR: arbitration lost
2018-10-05 16:50:54.650 [main error] scan config 15: ERR: wrong symbol received
2018-10-05 16:50:56.741 [main error] scan config 76: ERR: wrong symbol received
2018-10-05 16:51:13.645 [main error] scan config 08: ERR: arbitration lost
2018-10-05 16:51:14.776 [update notice] received unknown BC cmd: 10feb5160301a013
2018-10-05 16:51:15.762 [main error] scan config 15: ERR: wrong symbol received
2018-10-05 16:51:17.938 [main error] scan config 76: ERR: wrong symbol received `

WWebber, 2018-10-05:
ebusctl info shows this:

version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 42
max symbol rate: 127
min arbitration micros: 40
max arbitration micros: 75
reconnects: 0
masters: 4
messages: 14
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9

WWebber, 2018-10-05:
ebusctl scan / scan result always only reports "done".. nothing else..

WWebber, 2018-10-06:
a) how else could I check and read these ebusd.eu files (w/o starting the debugger)?
b) could I use an additional config path, so both paths are being processed?

WWebber, 2018-10-06:
sry about - opened another thread..

john30, 2018-10-07:
@semenoh another thing that just comes to my mind: is it possible that your PC running ebusd does not have internet connectivity all the time? or that DNS does not work some times?
it would be helpful to get your ebusd.log file after facing the signal lost problem

semenoh, 2018-12-16:
it looks like this issue I've managed to fix somehow...

closed 2018-12-16 by semenoh