john30/ebusd #205: Service timed out in munin

rchmd, 2018-10-04:
Hello,

After my SD crashed on the three-year-old RaspberryPi, I've now installed Raspbian Stretch and redesigned EBUSD. eBUS coupler Ethernet works great. EBUSD is going great.

Munin is also installed and the ebusd plugin is installed. Munin, however, reports the error "Service 'ebusd_power' timed out." (and all other ebus sensors) in /var/log/munin/munin-node.log

I have included other plugins without errors.

ebusd: 3.2 (build and installed in stretch)

I've changed in /etc/default/ebusd: EBUSD_OPTS="-d 192.168.178.33:5000 -s -p 8888 --loglevel=error"

root@raspberrypi:/home/pi# ebusctl i
version: ebusd 3.2.v3.2-11-g18bd21f
update check: revision v3.2-8-g0f77a9d available, broadcast.csv: newer version available, vaillant/08.ehp.csv: newer version available, vaillant/15.uih.csv: newer version available, vaillant/23.ehp.cc.csv: newer version available, vaillant/25.ehp.hwc.csv: newer version available, vaillant/50.ehp.mc.csv: newer version available, vaillant/broadcast.csv: newer version available, vaillant/errors.inc: newer version available, vaillant/hcmode.inc: newer version available, vaillant/iotesthp.inc: newer version available
signal: acquired
symbol rate: 41
max symbol rate: 134
min arbitration micros: 49
max arbitration micros: 367
min symbol latency: 8
max symbol latency: 18
reconnects: 0
masters: 3
messages: 617
conditional: 213
poll: 3
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"

Any ideas?

WWebber, 2018-10-05:
Quick question: which "eBUS coupler Ethernet" you use? I can't get my setup running.. It's not reporting CRC errors or similar, but my devices are not detected and I have the feeling that the CSV files on ebusd.eu are not even processed..

john30, 2018-10-05:
maybe munin is not allowed to connect to the port 8888, you might want to check that when running ebusctll with the user account that munin is running as.

rchmd, 2018-10-05:
@WWebber: https://www.esera.de/produkte/ebus/142/ebus-koppler-ethernet?c=20

rchmd, 2018-10-05:
Hello John,

thanks for your suggestion. Unfortunately, it works fine under the user munin:

`pi@raspberrypi:~ $ sudo -u munin ebusctl -p 8888
localhost: listen
listen started
ehp SourceTempInput = 20.19;ok
ehp Status = 21.25;1.551;1.225;heat;08`

`pi@raspberrypi:~ $ ps aux | grep munin
root       358  0.0  0.9  13588  9140 ?        Ss   Okt04   0:14 /usr/bin/perl -wT /usr/sbin/munin-node
munin    12918  0.0  0.0   1900   408 ?        Ss   21:25   0:00 /bin/sh -c if [ -x /usr/bin/munin-cron ]; then /usr/bin/munin-cron; fi
munin    12920  0.0  0.0   1900   420 ?        S    21:25   0:00 /bin/sh /usr/bin/munin-cron
munin    12922  2.0  1.9  59736 19612 ?        S    21:25   0:01 /usr/bin/perl /usr/share/munin/munin-update
munin    12927  0.2  1.7  60128 17080 ?        S    21:25   0:00 /usr/share/munin/munin-update [Munin::Master::UpdateWorker<localdomain;localhost.localdomain>] -- fetch(ebusd_power)
root     12928  0.1  0.8  13588  8452 ?        S    21:25   0:00 /usr/sbin/munin-node [::ffff:127.0.0.1]
root     13028  0.1  0.2   3464  2380 ?        Ss   21:26   0:00 /bin/bash /etc/munin/plugins/ebusd_power
root     13030  0.0  0.1   3460  1432 ?        S    21:26   0:00 /bin/bash /etc/munin/plugins/ebusd_power
pi       13035  0.0  0.0   4776   536 pts/0    S+   21:26   0:00 grep --color=auto munin
`
Should I going back to jessie?

WWebber, 2018-10-06:
> 
> 
> @WWebber: https://www.esera.de/produkte/ebus/142/ebus-koppler-ethernet?c=20

Dear rchmod, thank you - this is exactly the same I use, but I can't get it working with the CSV files.  Although it reads and shows all messages properly, it never decodes them to any cleartext.

john30, 2018-10-06:
@WWebber please don't mix completely different issues into one. I won't answer to yours unless you open a separate issue for that.

john30, 2018-10-06:
@rchmd maybe a different user account is used for that one, I don't know

rchmd, 2018-10-06:
Hello John. Thanks anyway. Thanks a lot for ebusd, a great tool. I'll get it working again for sure :-)

john30, 2018-10-07:
it seems that munin runs the query as user root, so you could check that this works as well. It might be just some simple path issue as well. you could go and run `/bin/bash /etc/munin/plugins/ebusd_power` as user root. it is not supposed to run into a timeout of any kind.
oh, maybe this was the initial run during powerup? and maybe ebusd was not ready at that time. just an idea...
anyway, I'll add a timeout to the nc calls in the munin script, at least it should no longer run into timeouts then

rchmd, 2018-10-07:
Hello John,
the result is:
root@raspberrypi:/bin/bash /etc/munin/plugins/ebusd_power
/etc/munin/plugins/ebusd_power: Zeile 6: /plugins/plugin.sh: Datei oder Verzeichnis nicht gefunden
ebusd_power_ehp_ActualEnvironmentPower.value 0

The same with "press" and "hours" --> Its working.

No, the issue is not only during powerup. But: I switched back to "Jessie" and ebusd is working fine with munin :-)

rchmd, 2018-10-07:
Hello John,
I was wrong. In my previous post I connected to "Jessie". With "Stretch" I got the timeout:

root@raspberrypi:/home/pi# /bin/bash /etc/munin/plugins/ebusd_hours
/etc/munin/plugins/ebusd_hours: Zeile 6: /plugins/plugin.sh: Datei oder Verzeichnis nicht gefunden

root@raspberrypi:/home/pi# /bin/bash /etc/munin/plugins/ebusd_power
/etc/munin/plugins/ebusd_power: Zeile 6: /plugins/plugin.sh: Datei oder Verzeichnis nicht gefunden

root@raspberrypi:/home/pi# /bin/bash /etc/munin/plugins/ebusd_press
/etc/munin/plugins/ebusd_press: Zeile 6: /plugins/plugin.sh: Datei oder Verzeichnis nicht gefunden

René

john30, 2018-10-08:
ok, still weird. did you try the latest update of [ebusd-munin](https://raw.githubusercontent.com/eBUS/ebusd-munin/master/ebusd_) ?

rchmd, 2018-10-08:
Hello John,
you got it! Great! It runs. 💯 
I reduced the timeout to one second and still works fine. THANK YOU.
René

john30, 2018-10-09:
great:)

closed 2018-10-09 by john30