john30/ebusd #206: Does not process CSV files ?

WWebber, 2018-10-06:
Using the esera ethernet module. 
ebusd does not "translate" messages to CSV definition files. Although it reads and shows all messages properly (w/o any CRC errors), it never decodes them to cleartext.
"ebusctl scan result" also shows always only "done".
No matter if run in foreground (-f option) or as service.

WWebber, 2018-10-06:
Thanks for the reply, but I'm sorry, I do not understand what you answered.
My issue with ebusd is, that I think the CSV files are not being processed. Neither with "scan results" nor with "find -d" is see any useful recognized and decoded informations.

john30, 2018-10-07:
your problem is the ethernet module used in the adapter. this is known to impose huge latency on the communication between ebusd and the bus through the used TCP/UDP to serial converter. this can be recognized from the [log you've provided in your previous post](https://github.com/john30/ebusd/issues/204#issuecomment-427393908):

> 2018-10-05 16:50:24.907 [main error] scan config 08: ERR: wrong symbol received
2018-10-05 16:50:34.340 [main error] scan config 15: ERR: arbitration lost
2018-10-05 16:50:36.614 [main error] scan config 76: ERR: wrong symbol received
2018-10-05 16:50:52.299 [main error] scan config 08: ERR: arbitration lost
2018-10-05 16:50:54.650 [main error] scan config 15: ERR: wrong symbol received
2018-10-05 16:50:56.741 [main error] scan config 76: ERR: wrong symbol received
2018-10-05 16:51:13.645 [main error] scan config 08: ERR: arbitration lost

these are all errors resulting from that and cannot be compensated. the best option you have if you want to stick to that adapter is using UDP as transport instead of TCP, but thats a little tricky to set up

WWebber, 2018-10-07:
I am not getting any of these arbitration lost errors. The module is working and all logfiles show no other warnings like "received unknown MS/BC commands". BTW: after playing around several hours, I accidentially found one way to make it working: When I reset my module (by pressing the <Setting> button in their ConfigTool) while ebusd is already up and running, ebusd recognizes this device reset and starts decoding immediately.

Dinth, 2018-10-07:
Ohh when i think that i was originally planning to buy this ethernet adapter for 150 pounds....
Thank God i found FHEM forums and their homemade RPi<->eBus modules :)

gperdach, 2018-10-15:
Hi WWebber,
I also have the ethernet adapter and I am always getting the arbitration lost & wrong symbol received errors.
how did you get rid of this problem?
Gregor

WWebber, 2018-10-19:
Did you already try to "restart" the adapter (press the <Setting> button in the ConfigTool), while ebusd is running? The reason might be that the adapter does not flush the data memory, when ebusd connects.

gperdach, 2018-10-24:
I changed the adapter-settings via ConfigTool to use UDP (see suggestion from @john30). Now it works!