john30/ebusd #207: write -h

WWebber, 2018-10-06:
Does not work at all - no matter what I output, always shows the help text. It's not documented, but is it also necessary to add "--enablehex" ?

WWebber, 2018-10-06:
Answering myself: I added "--enablehex", but it did not help. 
Does "write -h" also output the usage text, when just the passed byte string is faulty?

john30, 2018-10-07:
> Does "write -h" also output the usage text, when just the passed byte string is faulty?

yes

WWebber, 2018-10-07:
Then something in the code behaves different as the same string works perfect with the "hex" command.

john30, 2018-10-07:
you should really read [the wiki](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#write). "write -h ZZPBSBNN[DD]*" is only equal to "hex ZZPBSBNN[DD]*" if there is a message definition associated with the combination of source&destination address, PBSB and optionally parts of the master data.
that's why there is a dedicated hex command...

closed 2018-10-07 by john30

john30, 2018-10-08:
you certainly provided non-matching values if that is the case.
where exactly did you find those many examples?

TheEPIX, 2019-01-03:
I'm just a noob with ebus, but sometimes the fog ist clearing....
So I'm just trying to find out the communication of my WTC25 (weishaupt)
The actual problem is also the hex-writing:

I don't know how to use the "hex" command :-(

A example in Fhem uses
ebusctl write -h 0850030126 
to get some answers

i tried this and also 
ebusctl write -h hex0850030126 --enablehex
ebusctl write -h hex 0850030126 --enablehex
ebusctl write -h 0850030126 --enablehex
ebusctl write -hex0850030126 --enablehex
nothing bit the "usage-information" of ebusctl...

Any hint (if possibile "simple" for a noob)

john30, 2019-01-04:
duplicate of #248