john30/ebusd #208: Howto use -def with custom message definitions?

andig, 2018-10-09:
I've upgraded to 3.2 and would like to decode `PrimaryCircuitFlowRate` (d.029) which is shown on my BAI but not decoded. Since ebusd-config has both `UIN` and `uin100` I'd like to test if one of the two works (definition copied out of the config):

    ebusctl -s nas.fritz.box -p 8888 r -f -V -def 'r,,PrimaryCircuitFlowRate,d.29 PrimaryCircuitFlowRate_DK,,,,"FB00",,,uin100,,,'

Unfortunately all I'm getting is the usage instructions with message definition in hyphens or not. 

Would appreciate a hint ;)

john30, 2018-10-10:
hm, it seems that the space in the definition is problematic as the argument parsing will treat it as separator. needs to be fixed, but you can continue by using "d.29" instead of "d.29 PrimaryCircuitFlowRate_DK" for now

andig, 2018-10-10:
Still no luck. When I do that it complains about missing circuit which isn't part of the available options:

    pi@server:~ $ ebusctl -s nas.fritz.box -p 8888 r -f -V -def 'r,,PrimaryCircuitFlowRate,d.29,,,,"FB00",,,uin100,,,'
    ERR: bad definition: temporary:2: ERR: missing argument, circuit

    pi@server:~ $ ebusctl -s nas.fritz.box -p 8888 r
    usage: read [-f] [-m SECONDS] [-s QQ] [-d ZZ] [-c CIRCUIT] [-p PRIO] [-v|-V] [-n|-N] [-i VALUE[;VALUE]*] NAME [FIELD[.N]]
    or:  read [-f] [-m SECONDS] [-s QQ] [-d ZZ] [-v|-V] [-n|-N] [-i VALUE[;VALUE]*] -def DEFINITION
    or:  read [-f] [-m SECONDS] [-s QQ] [-c CIRCUIT] -h ZZPBSBNN[DD]*

Adding circuit `-c bai` then shows usage instructions again?

john30, 2018-10-10:
that's not possible since all the default stuff is no longer available

andig, 2018-10-10:
WIth the new `-def` feature I've also taken another look at https://github.com/john30/ebusd-configuration/issues/89 which confirmed that `PartloadHcKW` need be redefined for by bai:

    #r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"0704",,,power,,,Heizungsteillast
    r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"6C00",,,power,,,Heizungsteillast

While this used to work (which custom redefined messages):

    pi@ebus2:~ $ ebusctl r -V PartloadHcKW
    bai PartloadHcKW power=10 kW [Heizungsteillast]

this doesn't:

	pi@server:~ $ ebusctl -s nas.fritz.box r -def 'r;wi,BAI00,PartloadHcKW,d.00,,,"B509","6C00",,,power,,,Heizungsteillast'
	ERR: bad definition: temporary:2: ERR: element not found, field type POWER in field 0

User error or another issue with how `-def` works?

andig, 2018-10-11:
@john30 any insights on this: 

    pi@server:~ $ ebusctl -s nas.fritz.box r -def 'r;wi,BAI00,PartloadHcKW,d.00,,,"B509","6C00",,,power,,,Heizungsteillast'
    ERR: bad definition: temporary:2: ERR: element not found, field type POWER in field 0

definition working with custom csv file but not from command line. What am I doing wrong here?

john30, 2018-10-11:
with "-def" you need to fully specify everything, i.e. also field types completely, since no templates are loaded, no defaults are resolved, or anything like that.

andig, 2018-10-11:
Ahh, I see. Where would I find the definition of `POWER` in this example? It does not seem to be part of `_templates.csv`?

andig, 2018-10-11:
I'm still lost. Looking at https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/_templates.csv I would do it like this where power becomes uch:

    ebusctl -s nas.fritz.box r -def 'r;wi,BAI00,PartloadHcKW,d.00,,,"B509","6C00",,,UCH,,,Heizungsteillast'
    ERR: invalid address

...which still doesn't work?

john30, 2018-10-14:
as said before, you need to completely and fully put the whole definition in, basically the same what `ebuscl f -f` results in. So in the axample above, the destination address ZZ is missing.

andig, 2018-10-14:
> as said before, you need to completely and fully put the whole definition in

I've heard you but I didn't understand it, sigh :(. I thought I had put in the destination due to `BAI00`, but was missing the `ZZ="08"`. Working now, thanks for the lesson!

May be it would have been clearer if the error message was `invalid destination address`? It *sounded* like register address or whatever. If you don't want to change that happy to close!

john30, 2018-10-14:
> Working now, thanks for the lesson!

you're welcome!

> May be it would have been clearer if the error message was `invalid destination address`?

currently there is only one error message ID specified for all kinds of source or destination address problems. I'd rather keep it as it is, but it could be enhanced of course.

need to keep this bug open anyway, as the space problem still needs to be fixed

commit 7d10104d (2018-12-02): allow single quote for escaping as well (fixes #208)