john30/ebusd #209: ebusctl grab doesn't provide new messages

zach1000, 2018-10-10:
Hi

I have a problem with identifing new messages with ebusctl grab. (ebusd 3.2, Bartl Heating Pump with TEM Controller, ebus coupler USB, RasberryPI). The grab doesn't provide the messages in the format:

1023b5040101 / 091e0300000006000100 = 1
1023b504010d / 050000008000 = 1
1025b5040101 / 09350300000003000100 = 10
1025b504010d / 050000290135 = 10
1050b5040101 / 09150300000081000100 = 6
1050b504010d / 051700710115 = 6

Just for a short reminder, here is the meaning of the bytes shown above:
`QQZZPBSBNNDD / NNDD = count`

I get only the following format:

`pi@raspberrypi ~ $ ebusctl grab result all
10fe070009b31431101912100518 = 2: broadcast datetime
10fe0801084545454500004545 = 2: broadcast temperatures
1003050a00 = 1
1013050a00 = 1
3108070400 / 0a1057455f312036323030 = 1: scan.08
3118070400 / 0a1057455f322036323030 = 1: scan.18
3135070400 / 0a10313834373301090000 = 1: scan.35
100305020101 = 1
101305020101 = 1
10fe0802080000003000000000 = 3
10fe100a0e10030000000000000000e0016400 = 7
10fe100a0c11030000fe1a040400213500 = 7
100305010a00000015000000000000 = 4
101305010a00000015000000000100 = 4
30100800080000b3140000000a = 1
031005030a010000004a23ff3f0000 = 9
131005030a011a00004affff3f4300 = 9`

Also when I open the values in the controller display no new messages are shown.

`pi@raspberrypi ~ $ ebusctl info
version: ebusd 3.2.v3.2-11-g18bd21f
update check: revision v3.2-8-g0f77a9d available, broadcast.csv: different version available
signal: acquired
symbol rate: 47
max symbol rate: 73
min arbitration micros: 816
max arbitration micros: 949
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 280
conditional: 194
poll: 0
update: 2
address 03: master #11
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=20097;SW=0375;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=TEM;ID=18473;SW=0109;HW=0000"
address 36: slave #8, ebusd`

Does anybody have a idea why this happens or has a hint how to determine new messages?

Details (conf, logs, etc.) can be found here as well: https://bitbucket.org/silverstone1000/ebusd/src/master/

BR
Mario

john30, 2018-10-14:
maybe the controller just doesn't send/query everything it displays on/from the bus.
I'm not an expert for TEM stuff, so I have no idea

rellla, 2018-12-13:
hi @john30,
is it wanted, that "ebusctl grab" doesn't grab messages which are written or read with ebusctl directly or via fhem/ecmddevice? reading and writing itself works nice and the messages between the other masters/slaves are grabbed. i also have a tem controller btw.
thanks rellla

john30, 2018-12-22:
line 1076 does not have anything to do with that

rellla, 2018-12-22:
It has. Dropping line 1076 and 1078 and so always addseenaddress lets me see the messages from ebusd itself.
rellla

john30, 2018-12-25:
which address are you using for ebusd and which slave addresses are seen on the bus? to me it looks like this is an issue with your ebusd device address being set to one of the addresses uses by another device on the bus. check the output of "ebusctl info"

rellla, 2018-12-25:
Hi, this my ebusctl info:
```
epi@smartmeter:~ $ ebusctl info
version: ebusd 3.2.v3.2-23-g768bd5c
update check: revision v3.2-25-gc475e6c available, broadcast.csv: different version available
signal: acquired
symbol rate: 59
max symbol rate: 171
min arbitration micros: 964
max arbitration micros: 9874
min symbol latency: 0
max symbol latency: 28
reconnects: 0
masters: 5
messages: 292
conditional: 196
poll: 1
update: 5
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=TEM;ID=22273;SW=0160;HW=0000"
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=22090;SW=0521;HW=0102", loaded "tem/15.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
It's address is 31 and 36. What should it be?
Thanks rellla

PS: Great piece of software btw - if i didn't mention that already ;)