john30/ebusd #210: Howto write installer registers on ecoTEC plus?

andig, 2018-10-20:
To improve the behaviour of my VRC700 I need more control about desired values, `FlowsetHcMax` being one that appears easier to modify. Unfortunately, any write access is without effect (access level is granted):

	ebusctl -s nas.fritz.box r FlowsetHcMax
	64

	ebusctl -s nas.fritz.box w -def 'wi,BAI00,FlowsetHcMax,,,"08","B509","0e0e04",,,uin,,,' 60
	done

	ebusctl -s nas.fritz.box r FlowsetHcMax
	64

Here's the corresponding bus log:

	2018-10-20 09:40:14.895 [bus notice] >3108b509030d0e046c<00024003e7>00
	2018-10-20 09:41:43.629 [bus notice] >3108b509050e0e043c0028<000000>00

Now, I've tried to use VRC700 as sender address `15` instead, none of which is working:

	ebusctl -s nas.fritz.box w -s 15 -def 'wi,BAI00,FlowsetHcMax,,,"08","B509","0e0e04",,,uin,,,' 60
	ERR: invalid address

	ebusctl -s nas.fritz.box w -def 'wi,BAI00,FlowsetHcMax,,"15","08","B509","0e0e04",,,uin,,,' 60
	ERR: bad definition: temporary:2: ERR: invalid address, qq 15

I would appreciate and hint how to further tweak sending `wi` writes to the bus.

andig, 2018-10-20:
I've tried all master addresses using `-s <id>`, to no avail. Write reports done but `FlowsetHcMax` isn't changed. Retried after https://github.com/john30/ebusd-configuration/pull/107 was merged and active.

Is there any known secret to writing installer messages or is my bai just too stubborn?

john30, 2018-10-20:
maybe the definition isn't right. the data is a "temp" field based on D2C according to [bai.0010015600.inc](https://github.com/john30/ebusd-configuration/blob/5aa359bc44e92d4da2b8306841160f65556d01f3/ebusd-2.1.x/de/vaillant/bai.0010015600.inc#L103), so it consists of 2 bytes and the value is signed and divided by 16. so for an actual temperature of 60 degrees the value in UIN notation is 960.
the bai knows for each register the valid value range and will probably recognize that 60 (as UIN) is out of range.

andig, 2018-10-20:
Not sure what you mean. Its properly decoded to the expected value when reading. Same problem after using the updated ecotec file, i.e. without custom definition. Also same problem for basically and installer value I‘ve tried. It looks as if the bai just won‘t let me in. Maybe ˋ0eˋ is the wrong value for installer writes?

john30, 2018-10-24:
yes thats an option. please post the result of `ebusctl f -f -a FlowsetHcMax`

john30, 2018-10-25:
ok then I don't understand why you were using UIN as data type during the write test. as mentioned before, the value range is totally different with these two types

andig, 2018-10-25:
> I can read basically all installer parameters and the values appear plausible.

I have to correct myself. I've found `ReturnRegulation` which is `onoff` and I can write it! So its not as if the bai would prevent this in principle.

andig, 2018-10-30:
ok, I've tried something very stupid:

	for i in {0..255};do echo "$i: " $(ebusctl -s nas.fritz.box w -c bai partloadhckw $i >/dev/null && ebusctl -s nas.fritz.box r -f partloadhckw); done

The result was always 3 (which is the limit set in the bai) or $i. The funny thing is this:

[image]

It looks as if the max power (violet) actually *was* changed (the orange temperature followed) though it could not (immediately) be read?

john30, 2018-10-31:
maybe it writes to internal storage after a certain delay only, so you could try with waiting for lets say 10 or 30 seconds before reading again

andig, 2019-11-04:
Closing this one as no longer an issue for me. Instead of fiddling with the `BAI` controls, I've fixed the physical problems in the heating installation and do no longer need this level of control.

closed 2019-11-04 by andig

ooghes, 2024-09-28:
Hi all, I tought I'd share my experience with the above issue. I too wanted to configure the max flow temperature on my Vaillant ecoTEC plus boiler. My boiler type didn't have a matching hardware/software version, so the config files did a fallback to "bai.308523.inc", just like in the above comments. In this file, FlowsetHcMax is defined at address **0x0E04**.
Using that address, I got the very same phenomenon:

```
ebusctl r -c bai r FlowsetHcMax
60
ebusctl w -c bai FlowsetHcMax 55
done
ebusctl r -c bai r FlowsetHcMax
60
```
So one can read, but not write.

I started playing around with the other bai.xxx.inc configuration files and noticed that in some of them, a different address is used, i.e. 0xA500. When testing my configuration with that address, I was successful!

```
ebusctl r -c bai r FlowsetHcMax
60
ebusctl w -c bai FlowsetHcMax 55
done
ebusctl r -c bai r FlowsetHcMax
55
```

Here's the parameter definition that worked for me:

`r1;wi,,FlowsetHcMax,d.71 Max flow temperature,,,,"A500",,,temp,,,Adjustment of maximum central heating set point range (maximum end position of potentiometer)
`

(note: I changed to "r1" because I wanted to read out this parameter more frequently.