john30/ebusd #211: Custom message definitions not working through MQTT

andig, 2018-10-20:
Follow-up of #208. Once a custom message definition has been tested and declared via `define` it should be possible to read/write it through MQTT, too. Currently that doesn't seem possible- forced reads through MQTT `/get` do not appear on the bus and no result is received.

andig, 2018-10-20:
Before https://github.com/john30/ebusd-configuration/issues/107 was merged I didn't have `PrimaryCircuitFlowrate`. So I did:

	ebusctl -s nas.fritz.box define -r 'r,BAI00,PrimaryCircuitFlowrate,,,"08","B509","0Dfb00",,,uin,,,'

and used

	ebusd/bai/PrimaryCircuitFlowrate/get

for reading. That never got any response as if new definitions are not exposed on MQTT.

john30, 2018-10-20:
thats because you are using "BAI00" instead of "bai" as circuit in the definition

closed 2018-10-20 by john30

john30, 2018-10-28:
you can easily see the message definitions that ebusd compiles from templates and defaults and all that stuff by using "find -f -a".

john30, 2018-10-28:
hint added to the wiki