john30/ebusd #212: compilation on OSX

Mathadon, 2018-10-21:
Hi,

I tried to compile ebusd on OSX by following the instructions. I get the following error when running `make`:

```
tcpsocket.cpp:86:37: error: use of undeclared identifier 'errno'
    if (ret < 0 && (timeout <= 0 || errno != EINPROGRESS)) {
                                    ^
tcpsocket.cpp:86:46: error: use of undeclared identifier 'EINPROGRESS'
    if (ret < 0 && (timeout <= 0 || errno != EINPROGRESS)) {
                                             ^
```

is this a coding error or an error on my side? Thanks!

john30, 2018-10-21:
maybe adding errno.h to the includes might help

Mathadon, 2018-10-21:
```
git diff
diff --git a/src/lib/utils/tcpsocket.cpp b/src/lib/utils/tcpsocket.cpp
index f6a752d..d33dc70 100755
--- a/src/lib/utils/tcpsocket.cpp
+++ b/src/lib/utils/tcpsocket.cpp
@@ -25,6 +25,7 @@
 #include <arpa/inet.h>
 #include <netdb.h>
 #include <string.h>
+#include <errno.h>
 #ifdef HAVE_PPOLL
 #  include <poll.h>
 #endif
```

indeed fixes it!

commit 45b9bad5 (2018-10-21): missing include (fixes #212)
closed 2018-10-21 by john30