john30/ebusd #213: "unable to open /dev/ttyUSB0"

Mathadon, 2018-10-21:
I just started to use ebusd. Managed to get it compiled on Ubuntu 14.04 (in VM) and hooked up an Esera USB koppler, which is connected to my thermostat (Bulex exacontrol). I turned the potentiometer until the 'Data' LED stays on. I tried to verify the connection using the command from the wiki/howto, which results in:

```
ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0 
2018-10-21 14:30:12.278 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2018-10-21 14:30:12.278 [bus notice] bus started with own address 31/36
2018-10-21 14:30:12.278 [bus notice] device invalid
2018-10-21 14:30:17.279 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2018-10-21 14:30:17.279 [bus notice] device invalid
2018-10-21 14:30:22.279 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2018-10-21 14:30:22.280 [bus notice] device invalid
etc
```

some more info:

```
ls -la /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Oct 21 14:31 /dev/ttyUSB0
```

```
dmesg | grep usb
[ 2316.008931] usb 2-1: USB disconnect, device number 4
[ 2328.521023] usb 2-1: new full-speed USB device number 5 using uhci_hcd
[ 2328.662656] usb 2-1: New USB device found, idVendor=0403, idProduct=6001
[ 2328.662660] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2328.662663] usb 2-1: Product: eBus Coupler Iso 12001
[ 2328.662665] usb 2-1: Manufacturer: ESERA-Automation
[ 2328.662666] usb 2-1: SerialNumber: AL2I7EJY
[ 2328.733151] usb 2-1: Detected FT232RL
[ 2328.733154] usb 2-1: Number of endpoints 2
[ 2328.733157] usb 2-1: Endpoint 1 MaxPacketSize 64
[ 2328.733160] usb 2-1: Endpoint 2 MaxPacketSize 64
[ 2328.733162] usb 2-1: Setting MaxPacketSize 64
[ 2328.736174] usb 2-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

Possibly relevant: running this Ubuntu 14.04 in a VM on osx. Using a USB-C to USB converter to connect the esera.

Any tips on how to debug this? Thanks a lot!

john30, 2018-10-21:
did you run ebusd as root as well? sounds like a rights problem to me

Mathadon, 2018-10-21:
Looks related, now I get: 

```
sudo ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0 
[sudo] password for parallels: 
2018-10-21 14:49:27.789 [bus notice] bus started with own address 31/36
```
then nothing for at least one minute.

I also tried directly on OSX now that compilation was fixed in #212 : 
```
sudo ./ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/tty.usbserial-AL2I7EJY
```
this returns nothing at all for at least one minute.

In both cases, unplugging the USB results in more
```
2018-10-21 14:51:32.972 [bus notice] device invalid
2018-10-21 14:51:37.973 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
``` 
so the process did not crash.