john30/ebusd #214: SetMode (passive write) not visible on MQTT

andig, 2018-10-31:
I'm trying to listen to `SetMode` messages using MQTT. I can see the messages are generated every 10 seconds:

    $ watch -n 10 --difference 'ebusctl -s nas.fritz.box grab result all decode | grep -a -A 0 1008b510'

increases the msg counter every time. Yet, the mosquitto topic is only updated every couple of minutes:

    $ mosquitto_sub -h nas.fritz.box -t ebusd/bai/SetMode -v

All other BAI messages ('Status01' etc) are visible:

	$ mosquitto_sub -h nas.fritz.box -t ebusd/bai/# -v
	ebusd/bai/PrimaryCircuitFlowrate 16.71
	ebusd/bai/ReturnRegulation off
	ebusd/bai/ExtFlowTempDesiredMin 40.00
	ebusd/bai/PrimaryCircuitFlowrate 16.71
	ebusd/bai/Status01 33.5;31.0;11.000;-;56.0;on

...only `Status01` seems to appear only when changed, too.
 
It would be helpful if *all* messages would be visible on MQTT regardless of passive writes or caching.

john30, 2018-10-31:
it is only updated in MQTT when the value(s) change

closed 2018-10-31 by john30

andig, 2018-10-31:
If you're saying that is how MQTT behaves then this is not true. See below what happens when publishing the identical message twice:

    ❯ mosquitto_sub -h nas.fritz.box -t test -v
    test foo
    test foo

Every single message is relayed to the clients- MQTT does not deduplicate!

john30, 2018-10-31:
right, but ebusd does, since there is no benefit in sending the same thing all the time IMO.

andig, 2018-10-31:
My use case is this:

  - have a custom definition `SetModeOverride` that is writable
  - write to `SetModeOverride`
  - wait for `SetMode` sent by VRC700
  - repeat overriding that value with `SetModeOverride`

This is only possible when I see `SetMode` messages which I don't since they're hidden by ebusd as it does not recognize that `SetModeOverride` applies to `SetMode`, too.

andig, 2018-11-04:
> see my comment in #211

Spot-on as always. Sigh.

> furthermore, you cannot expect ebusd to update 2 message definitions from a single message on a bus. it is always 1:1

Yes, of course. I still think that the deduplication that `ebusd` does is debatable. For example, think about `SetMode` and `FlowTempDesired`- these two might really address the same register. Writing `SetMode` twice with the same value may well be more than a nop operation.

Would it make sense to:

  a) not cache writes as it is not clear that they even set a register to the desired value or
  b) add something like an `mqttverbatim` option that allows deactiviating the deduplication behaviour?

john30, 2018-11-04:
hm, i'm not sure that you've understood completely. or maybe i don't get your point.
with "deduplication", what exactly do you mean?
from ebusd MQTT point of view, it is no deduplication but simply a notify-on-change behaviour, i.e. an incoming message with exactly the same values as in the last transmission will not lead to an MQTT message.
it seems to me that you'd like ebusd to do a 1:N multiplication (which it won't), but as said, I'm not entirely sure what you mean.

andig, 2018-11-04:
> hm, i'm not sure that you've understood completely. or maybe i don't get your point.
with "deduplication", what exactly do you mean?

I mean the notify-on-change behaviour- it's essentially deduplicating messages of identical content.

> an incoming message with exactly the same values as in the last transmission will not lead to an MQTT message.

Yes. I'm questioning if this is always a good idea.

> it seems to me that you'd like ebusd to do a 1:N multiplication (which it won't).

No, I understand that makes little sense. It is already the case though that 2 different writes can update the same target register. Both `SetMode` and `FlowTempDesired` update `FlowTempDesired`. Imagine this sequence:

  1. write `SetMode=auto,35,....` -> triggers MQTT message
  2. write `FlowTempDesired=40` -> triggers MQTT message
  3. write `SetMode=auto,35,....`

The last write does *not* trigger a message although it does change `FlowTempDesired`!

I could also imagine (though I don't have an example) that a controller might open a valve on the BAI and the BAI would decide to close it, maybe based on temperature. Opening the valve a second time from the controller would not be visible on MQTT event though the valve state changes.

john30, 2018-11-07:
agreed, but also ebusd would not have any knowledge of no longer up-to-date infos in this particular case. So this is more a general question IMO. anyway, I've added #219 for forcing MQTT publish.