john30/ebusd #216: .deb release for Raspbian Stretch

mivola, 2018-11-01:
Hi,

is there a plan to offer a binary release for Raspbian Strech on RaspberryPi/armhf? Or should the "ebusd-3.2_armhf-jessie.deb" be sufficient?

Thanks,
Michael

john30, 2018-11-02:
at some point, yes. gor the time being, the jessie release should potentially work as well

mivola, 2018-11-02:
Thank you for the quick response. I'll give it a try ...

mivola, 2018-11-02:
I just did the installation and it seems to be working fine. Now I just need to understand & get the new ebus-configuration up & running ...

closed 2018-11-02 by mivola

Dinth, 2018-11-02:
>is there a plan to offer a binary release for Raspbian Strech on RaspberryPi/armhf? Or should the "ebusd-3.2_armhf-jessie.deb" be sufficient?
Ive been using it for a while without any problem