john30/ebusd #217: [main error] scan config 08: ERR: read timeout

gte477k, 2018-11-02:
Hi, 
I am starting to use ebusd to read values from my Vaillant ecoTec exclusive VC 156/5-7 with a VRC700. I am using ebusd 3.2.p20181101 compiled on an Ubuntu bash (within a W10 environment). Ebus communication works, DNS/Internet access is available as well. Checkconfig seems to work (however, only 11 messages):

ebusd --checkconfig --configpath=http://ebusd.eu/config/ --scanconfig=full -f -d /dev/ttyS4
2018-11-02 17:37:16.202 [main notice] ebusd 3.2.p20181101 performing configuration check...
2018-11-02 17:37:16.479 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2018-11-02 17:37:16.480 [main notice] ebusd stopped

When I am starting the deamon, I get ERR: read timeout for all messages:

ebusd --configpath=http://ebusd.eu/config/ --scanconfig=full -f -d /dev
/ttyS4
2018-11-02 17:34:09.795 [main notice] ebusd 3.2.p20181101 started with full scan
2018-11-02 17:34:09.997 [bus notice] bus started with own address 31/36
2018-11-02 17:34:10.042 [bus notice] signal acquired
2018-11-02 17:34:10.154 [bus notice] new master 10, master count 2
2018-11-02 17:34:10.218 [bus notice] new master 03, master count 3
2018-11-02 17:34:10.218 [update notice] received unknown MS cmd: 1008b5110101 / 095255000bff5e0000ff
2018-11-02 17:34:10.473 [update notice] received unknown MS cmd: 1008b51009000059ffffff000000 / 0101
2018-11-02 17:34:19.998 [main notice] starting initial full scan
2018-11-02 17:34:25.095 [main error] scan config 08: ERR: read timeout
2018-11-02 17:34:27.205 [main error] scan config 15: ERR: read timeout

My questions now is: is there a problem with the http://ebusd.eu/config/, or am I doing something wrong. I have tried it last year already, and some messages could be decoded at that time (08;Vaillant;BAI00;0502;9102, 15;Vaillant;70000;0419;4603).

Many thanks and kind regards,

Jens

john30, 2018-11-04:
what kind of interface are you using?

gte477k, 2018-11-04:
Hi, the Ebusd Circuit Board from Benedikt Patt.

john30, 2018-11-04:
can't tell from the input yet, I would need some raw logging from ebusd log, i.e. enable raw logging when starting ebusd like that:
`ebusd --lograwdata --configpath=http://ebusd.eu/config/ --scanconfig=full -f -d /dev/ttyS4`

john30, 2018-11-04:
your device just cannot write to the bus, so scanning is not possible and neither is actively querying something from another device

gte477k, 2018-11-05:
John, sorry to bother you, but it would be great if you could help me out:

-          Is my hardware adapter not able to write to the bus?

o   If not, which adapter would you recommend?

-          Or is my tty device not able to write to the bus?

o   Then, I would need to check the Ubuntu under W10 (or try a different OS setup). 

-          Which message tells me, that it cannot write to the bus? 

o   If I have changed my setupt, which message tells me, that I then can write to the bus?

Many thanks and kind regards,

 

Jens

john30, 2018-11-07:
> Is my hardware adapter not able to write to the bus?

you see that in the log as sent bytes (e.g. > 31) are never received back (missing < 31)

> If not, which adapter would you recommend?

see the [wiki](https://github.com/john30/ebusd/wiki)

> Or is my tty device not able to write to the bus?

I don't know.

> Which message tells me, that it cannot write to the bus?

see above

> If I have changed my setupt, which message tells me, that I then can write to the bus?

see above