john30/ebusd #218: "invalid device" after upgrade to v3.2

stefaanv, 2018-11-03:
Hi,
I had a rather successfull deployment of ebusd but decided to upgrade from 3.0 to 3.2 because I had a problem writing the `Hc1OPMode` parameter.
I installed the `ebusd-3.2_amd64-stretch_mqtt1.deb` package which I downloaded from the [ebusd.eu] 
 website.
However, after the upgrade ebusd is unable to communicate with my esera eBUS Koppler Ethernet while nothing changed at the koppler side.  The error message is listed below.
ebusd is running on an AMD64 computer (an intel NUC) with Ubuntu 16.04 Xenial operating system. Pinging the esera device from the computer works fine.
Running the same command with sudo has the same effect.

Any ideas on this ?

```
$ebusd --scanconfig -f -d 192.168.0.36:5000
2018-11-03 19:43:37.972 [main notice] ebusd 3.2.v3.2 started with auto scan
2018-11-03 19:43:38.221 [bus error] unable to open 192.168.0.36:5000: ERR: generic I/O error
2018-11-03 19:43:38.222 [bus notice] bus started with own address 31/36
2018-11-03 19:43:38.222 [bus notice] device invalid
2018-11-03 19:43:43.223 [bus error] unable to open 192.168.0.36:5000: ERR: generic I/O error
2018-11-03 19:43:43.223 [bus notice] device invalid

$ ping 192.168.0.36
PING 192.168.0.36 (192.168.0.36) 56(84) bytes of data.
64 bytes from 192.168.0.36: icmp_seq=1 ttl=128 time=0.223 ms
64 bytes from 192.168.0.36: icmp_seq=2 ttl=128 time=0.449 ms
```

stefaanv, 2018-11-03:
I tried to downgrade back to V3.0, but this isn't working
```
$ sudo apt install ./ebusd-3.0_amd64_mqtt1.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'ebusd' instead of './ebusd-3.0_amd64_mqtt1.deb'
The following packages will be DOWNGRADED:
  ebusd
0 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Need to get 0 B/211 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 /home/stefaan/ebusd-3.0_amd64_mqtt1.deb ebusd amd64 3.0 [211 kB]
dpkg: warning: downgrading ebusd from 3.2 to 3.0
(Reading database ... 148641 files and directories currently installed.)
Preparing to unpack .../ebusd-3.0_amd64_mqtt1.deb ...
Unpacking ebusd (3.0) over (3.2) ...
dpkg: error processing archive /home/stefaan/ebusd-3.0_amd64_mqtt1.deb (--unpack):
 trying to overwrite '/etc/ebusd/broadcast.csv', which is also in package ebusd-configuration 2.1.b143f39-en
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Start the daemon with 'systemctl start ebusd'
3. Check the log file /var/log/ebusd.log
4. Make the daemon autostart with 'systemctl enable ebusd'
Errors were encountered while processing:
 /home/stefaan/ebusd-3.0_amd64_mqtt1.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
stefaan@home:~$ ebusd -V
ebusd 3.2.v3.2
```

andig, 2018-11-03:
Instead ofdowngrading you could try the docker image. As you‘re using tcp- does latency make a change for you?

john30, 2018-11-04:
for the downgrade: since you've also installed the ebusd-configuration package forced ("-f"), you consequently also have to install ebusd in the same way (using "-f").
I can only assume that another ebusd instance was still running when you did the downgrade. just check your process list

stefaanv, 2018-11-04:
Solved.
I uninstalled and purged both ebusd-configuration and ebusd packages, rebooted the machine and reinstalled ebusd v3.2
Now setting to the Hc1OPMode parameter also works.
Thanks for the help !

closed 2018-11-04 by stefaanv