john30/ebusd #219: add --mqttforce

john30, 2018-11-07:
to always publish an MQTT message even if the message content did not change at all

andig, 2018-11-26:
For sake of completeness: this option is also needed if you want to do proper logging:

[image]

The yelllow solid line should in reality follow the dashed line. It doesn't since- without this option- ebusd silently ignores the repeated values. In the end, the logged temperature curve is therefore wrong.

@john30 might even make sense to swap the behaviour, i.e. make this option the default?

Dinth, 2018-11-26:
Great news! I hope this will also be a workaround for my problem #222

andig, 2018-11-27:
Would be great to get this into 3.3, I'm really looking forward to monitor my heating in more detail...

closed 2018-12-02 by john30