john30/ebusd #220: Community discussion forum

Dinth, 2018-11-07:
This is not an issue with eBus, but same can be said about many "issues" here which are just general questions. Maybe it would be worth to create a discussion forum on ebusd website, where users could ask generic questions, discuss their ebus configurations, etc? This would potentially decrease the time-cost of maintaining eBus software (as @john30 wouldnt need to read & tag all the discussions there and also, it would give users a place to discuss eBus, share ideas, etc without need of opening Github issue on a subject which is not an issue with the daemon

Dinth, 2018-11-07:
That's indeed one of the solutions, as FHEM members contribution into ebusd was quite large (eBus RPi HATs, eBus Home Assistant component), but it also has two downsides:
1. It's mostly related to one of the HA systems (FHEM) and it wont be an intuitive place to look for ebusd information fore OpenHAB or HA users.
2. It's mostly german-speaking 
But you're right, running a self hosted forum is not an option, becasue it would be time consuming for @john30.

Maybe it would be better to set up a forum hosted on Disqus infrastructure (just look on HA forums - i believe that those are hosted on Disqus too)?

Dinth, 2019-01-07:
What about Wikia? Its free, it's hosted, it's popular and it's in a form of knowledge base where everyone can contribute. 
And apparently it's really easy to export a wiki from Wikia if in the future if @john30 would like self-host.

john30, 2019-01-08:
îsn't wikia more a wiki page than a discussion platform?

Dinth, 2019-01-08:
Indeed Wikia allows creation of mediawiki-based encyclopedias, but every "knowledge" page also has a discussion page. Just as an example, it could host pages for common boilers and then if the page doesnt answer someones question, that person could always ask it on boilers discussion page.

john30, 2019-01-13:
I've added discus to ebusd.eu to start with, see here: https://ebusd.eu/discussion/

zyzakin, 2022-01-11:
Hello.
I have a Vaillant vc-206 / 5-5 R6 ecotec plus stove. 
The problem is with reading the values ​​of the PrEnergySumHc1 and PrEnergyCountHc1 keys. the point is, the value is constant (4,294,967,237.0) and never changes. I don't know if it's a problem with the stove or with the csv file.

Command: ebusctl grab result all gives the result: 
3108b509030df500 / 04c5ffffff = 234: bai PrEnergySumHc1
3108b509030df600 / 0447036401 = 237: bai PrEnergyCountHc1

```
ebusctl info
version: ebusd 21.2.v21.2
update check: version 21.3 available
signal: acquired
symbol rate: 77
max symbol rate: 136
min arbitration micros: 1
max arbitration micros: 4454
min latency symbol: 0
max latency symbol: 41
reconnects: 1
masters: 2
messages: 224
conditional: 3
poll: 0
update: 10
address 03: master # 11
address 08: slave # 11, scanned "MF = Vaillant; ID = BAI00; SW = 0104; HW = 7803", loaded "vaillant / bai.308523.inc", "vaillant / 08.bai.csv"
address 31: master # 8, ebusd
address 36: slave # 8, ebusd
```

I would like to add that it also has a 700 room regulator, which does not seem to be recognized.

My adapter is http://blog-techniczny.pl/wp-content/uploads/2017/09/ebus-ttl-converter.jpg

Has anyone had a similar problem and was it solved?

Greetings

kalledausb, 2022-01-14:
Hi,
I have an ecotec plus as well and I do get updates of these values if I force an update (ebusctl r -f -c bai PrEnergySumHc1).
My (in device integrated) 700 is recognized without issues as well, no idea what's the problem there.

john30, 2022-09-18:
[gh discussions](https://github.com/john30/ebusd/discussions) is the choice for now

closed 2022-09-18 by john30