john30/ebusd #222: MQTT weird behaviour

Dinth, 2018-11-11:
I noticed a weird behaviour in ebusd, if the ebusd daemon goes online sooner than MQTT server (Mosquitto).
I've got a script which periodically sends "1" to #/get of all the variables im interested in. Normally this works perfectly fine, ebusd updates all the variables after receiving a message to #/get. But sometimes, if ebusd is switched on before Mosquitto server (it happens quite often after power failures in my setup, as RPi boots faster than a full fledged server), ebusd still speaks with MQTT (sends "uptime" updates, periodically a single variable gets updated), but doesnt react on messages sent to #/get to force variable update.

```
ebusd/global/uptime 92626
ebusd/global/uptime 92642
ebusd/bai/ReturnTemp/get 1
ebusd/bai/ReturnTempExternal/get 1
ebusd/bai/WaterPressure/get 1
ebusd/bai/FlowTemp/get 1
ebusd/bai/FlowTempDesired/get 1
ebusd/bai/HwcTemp/get 1
ebusd/bai/HwcTempMax/get 1
ebusd/bai/HwcWaterflowMax/get 1
ebusd/bai/PositionValveSet/get 1
ebusd/bai/PartloadHcKW/get 1
ebusd/bai/ModulationTempDesired/get 1
ebusd/bai/PumpPower/get 1
ebusd/global/uptime 92658
ebusd/global/uptime 92674
ebusd/global/uptime 92690
```

Dinth, 2018-11-15:
Just used the different Mosquitto shocker container to be sure it’s not MQTT server issue, but I’m getting same results:
1) If Mosquitto server starts earlier than ebusd then everything works fine;
2) If ebusd starts earlier than Mosquitto server it still connects to it and publishes MQTT messages, but posting a message to ebusd/bai/variable/get doesn’t refresh the variable. When ebusd is restarted, everything goes back to normal

bo00, 2018-11-21:
I've a similar behaviour but independent to the starting time of Mosquitto.
Polling ebusd through /get messages works good but after some time ebusd publishes global/running with value "false". Afterwards it continues publishing global/uptime and some single variables but doesn't answer to /get requests.

Versions:
ebusd 3.2
mosquitto version 1.4.10

commit c9eae87b (2018-12-02): send version, running, and current signal after reconnect, re-set message callback on reconnect (fixes #222)
closed 2018-12-02 by john30