john30/ebusd #223: MQTT connect to Google Cloud IoT Core

teichsta, 2018-11-14:
Good Morning,

first things first: thanks a lot @john30 for this great piece of work!

Currently i am trying to connect my ebusd-mqtt to the Google IoT Core MQTT broker. I was able to connect to my Google IoT device from my Raspberry Pi using the given python sample (Paho Client, see https://github.com/GoogleCloudPlatform/python-docs-samples/blob/master/iot/api-client/mqtt_example/cloudiot_mqtt_example.py).

No i am trying to connect the ebusd unfortunately to no avail. There are some requirements, which i am not sure if they are currently available in ebusd-mqtt. E.g. The client-id needs to be set according to the "Google-Rules" (see https://cloud.google.com/iot/docs/how-tos/mqtt-bridge), which is possible to configure right now, correct?

Best, Thomas E.-E.

teichsta, 2018-11-14:
`EBUSD_OPTS="--loglevel=debug --configpath=/etc/ebusd --scanconfig \
   --mqttcert=/home/pi/certificates/rsa_private.pem \
   --mqtthost=mqtt.googleapis.com \
   --mqttport=8883 \
   --mqttjson \
   --mqttca=/home/pi/certificates/roots.pem \
   --mqttuser=unused \
   --mqttpass=eyJhbGciOiJSUzIdfgfdfas09438u \
   --mqtttopic=projects/project-281313/topics/ebusd`

i didn't find any option to set the client-id. Where can i do this?

andig, 2018-11-14:
> When the MQTT client connects, it must specify the device by setting the MQTT client ID to the full device path:
> i didn't find any option to set the client-id. Where can i do this?

Ouch. Looks like this is currently not possible: https://github.com/john30/ebusd/blob/671c8d2881a1c9bda599824d319c4dd4f4096c93/src/ebusd/mqtthandler.cpp#L349

cc @john30 could you kindly make `mqttclientid` configurable? I'd do a PR if I had any idea about C++...

teichsta, 2018-11-15:
@john30 i created PR #226 so "solve" this issue … hope this helps at least a little bit (as i am no C programmer :-))

john30, 2018-12-25:
setting the clientid is solved with 224c947499e9dbf2c98c890a87d69e942f84ae26
setting the protocol version to 3.1.1 is solved with cb78d981857d4315bfb3568ffcab1c0b1452f534

closed 2018-12-25 by john30