john30/ebusd #224: Feature request: MQTT Discovery

Dinth, 2018-11-14:
MQTT is not a very intuitive system and a lot of users struggle with setting it up (not only with ebus). There's a system which allows autodiscovery of MQTT services:

https://www.home-assistant.io/docs/mqtt/discovery/

I dont know C++ but i bet it wont be easy to implement this. This being said, i believe that this system would allow ebusd to grow it's userbase faster.

andig, 2018-11-14:
Looking at the link I'm assuming that you're suggesting that ebusd should support Homeassitant's proprietary device discovery protocol? 
Imho that is not core responsibility of ebusd but could be implemented outside of it, e.g. using the output of `ebusctl find` with the help of `mosquitto_pub`.

splatch, 2018-11-14:
@andig I think there is a universal way described in Homie convention: https://homieiot.github.io/.
As you can find there - it is supported in OpenHab 2.4 (to be released soon) as well as in Home Assistant. By following rules described above it will be possible to plug `ebusd` directly in any tool which understands appropriate topic names.

andig, 2018-11-14:
Isn't your link to hassio is a different thing from homie? The hassio link explains how to *tell* hassio which devices exist on mqtt while homie describes a self-describing mqtt topic hierarchy for compatible devices?

Or I'm getting this all wrong, sight.

Please don't get me wrong- I'm not debating this is isn't useful, I just don't think that its core ebusd responsibility.

Dinth, 2018-11-14:
No, you're right @andig, i never used Homie, but i was under impression that HA Discovery is just an implementation of Homie. This means that there are two very similar although different standards of discovery, both already quite well estabilished with hardware support, etc.
Anyhow, this is not a "core functionality" feature request, but just a "it would be nice to have it in the future" kind of request :)

andig, 2018-11-18:
Does anyone know of a comparison which discovery convention is "more" universal or has best support?

Dinth, 2018-11-18:
> Does anyone know of a comparison which discovery convention is "more" universal or has best support?

Both have some decent hardware support, but they differ with software support:
Home Assistant Discovery obviously works with HA.
Homie seems to have some initial OpenHAB support (but wait till someone confirms it) and i believe that Domoticz also supports it. 
Home Assistant Discovery should work almost out of the box though, while according to posts i found about Homie, it needs some tinkering to make it work even on systems which support it.

andig, 2018-12-02:
Btw, I have just implemented Homie for GoSDM in https://github.com/gonium/gosdm630/issues/85. Homie looks definitely interesting- it remains to be seen how far adoption will go. I.e. iobroker does not offer it yet although its conceptual approach seems quite close.

One interesting aspect for the implementation is that it makes sense- since homie devices are retained in mqtt- to unpublish the entire device tree upon startup in case structure has changed.

Dinth, 2019-12-25:
Seems that HA discovery gets a wider adoption so far - Sonoff switches, Valetudo (robotic vacuum open firmware), Zwave2MQTT, Zigbee2MQTT.

john30, 2019-12-25:
this is really not an easy one as each message would have to mapped to some HA device if I understood the documentation correctly. This is especially tricky as combined values (like temperature and status) as seen on eBUS can not be easily integrated like that. Somebody with deeper knowledge of HA or willing to spend a lot of time would be needed here. I don't have that time available.

Dinth, 2019-12-25:
Hi. You're right, HA would need to create a entity for each of the values, that's the point of autodiscovery. I think it would actually work great with eBus, as everybodys setup has different values, a lot of values, so manual creation of sensors for HA may take many, many hours and cannot be just copied and pasted to another setup. 
I wish i could help with that, but im a really lousy hello world level coder. Can help with translations, documentation or testing.

john30, 2022-03-19:
implemented

closed 2022-03-19 by john30