john30/ebusd #225: SyntaxError: JSON.parse: unexpected keyword at line 1379 column 25 of the JSON data

gte477k, 2018-11-14:
I have ebusd running, using Domoticz via JSON to communicate with the deamon. However, every once in a while the JSON file cannot be properly parsed anymore (error from Domoticz, same when using firefox):

SyntaxError: JSON.parse: unexpected keyword at line 1433 column 25 of the JSON data

When I check the JSON file is is as follows:

   "SystemFlowTemp": {
    "name": "SystemFlowTemp",
    "passive": false,
    "write": false,
    "lastup": 1542221911,
    "zz": 21,
    "id": [181, 36, 2, 0, 0, 0, 75, 0],
    "fields": {
     "tempv": {"value": nan}
    },

The error seems to result from the ebusd message:
2018-11-14 19:58:13.198 [update error] unable to parse read 700 z1Name2 from 0015b52406020003001800 / 06030318003100: ERR: invalid position

I am using the following config for 15.700
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.

The problem seems to lie in the "nan" value of the field. Do you have any idea on how to solve this?

Many thanks,

Jens

andig, 2018-11-14:
Nan does not exist in json. Encoder needs to respect this, maybe using null instead.

john30, 2018-12-02:
could you please post the raw data for SystemFlowTemp in this case?

closed 2018-12-02 by john30

gte477k, 2018-12-17:
Sorry, I am only looking at this now again. Thanks for implementing nan-support. Here you can find one "nan" message (z1RoomTemp):

root@ubuntu:/home/ubuntu/ebusd-configuration/latest/de/vaillant# cat  /var/log/ebusd.log | grep z1RoomTemp
2018-12-17 18:08:10.248 [update notice] received read 700 z1RoomTemp QQ=00: nan
root@ubuntu:/home/ubuntu/ebusd-configuration/latest/de/vaillant# ebusctl grab result all | grep z1RoomTemp
0015b52406020003000f00 / 0801030f00ffffff7f = 1: 700 z1RoomTemp
root@ubuntu:/home/ubuntu/ebusd-configuration/latest/de/vaillant#

I tried now to compile your update, I get the following error:

  CXX      datatype.o
datatype.cpp: In member function ‘virtual ebusd::result_t ebusd::NumberDataType::readSymbols(size_t, size_t, const ebusd::SymbolString&, ebusd::OutputFormat, std::ostream*) const’:
datatype.cpp:699:20: error: ‘isnan’ was not declared in this scope
       if (isnan(val)) {
                    ^
datatype.cpp:699:20: note: suggested alternative:
In file included from /usr/include/c++/5/random:38:0,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from ../../../src/lib/ebus/filereader.h:22,
                 from ../../../src/lib/ebus/datatype.h:32,
                 from datatype.cpp:23:
/usr/include/c++/5/cmath:641:5: note:   ‘std::isnan’
     isnan(_Tp __x)
     ^
Makefile:399: recipe for target 'datatype.o' failed. 

I am using Ubuntu 16.04.5 LTS (GNU/Linux 4.4.0-1100-raspi2 armv7l). 

Many thanks and kind regards

john30, 2018-12-22:
should work with be8d772524a82cd61bcd2fca2307de6a77f0f65e

gte477k, 2018-12-23:
Perfect. That worked. Many thanks and Merry Christmas.

john30, 2018-12-23:
great. thanks, same to you!

commit 8ed1e4f1 (2018-12-02): fixed NaN in JSON (fixes #225)