john30/ebusd #230: No boiler detected

Racailloux, 2018-12-15:
Hello,

1st of all, thanks for the great job (it looks to be great, because that doesn't work yet on our side :-)

We are using an eBUS adapter 2.2 (pre-soldered) connected to a R.Pi3 USB port. The eBus is connected to the 2-wire of a Vaillant EcoTec Plus.

The eBus seems to be communicating :

`pi@DomoPy:~ $ ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0
2018-12-15 22:31:48.998 [bus notice] bus started with own address 31/36
2018-12-15 22:31:49.052 [bus notice] <1e
2018-12-15 22:31:49.052 [bus notice] signal acquired
2018-12-15 22:31:49.056 [bus notice] <d1
2018-12-15 22:31:49.060 [bus notice] <8a
2018-12-15 22:31:49.065 [bus notice] <4c
2018-12-15 22:31:49.070 [bus notice] <39
2018-12-15 22:31:49.075 [bus notice] <48
2018-12-15 22:31:49.081 [bus notice] <65
2018-12-15 22:31:49.085 [bus notice] <b0
2018-12-15 22:31:49.089 [bus notice] <e8
2018-12-15 22:31:49.096 [bus notice] <d8
2018-12-15 22:31:49.099 [bus notice] <49
2018-12-15 22:31:49.106 [bus notice] <59
2018-12-15 22:31:49.198 [bus notice] <04
2018-12-15 22:31:49.204 [bus notice] <c2
2018-12-15 22:31:49.209 [bus notice] <84
2018-12-15 22:31:49.214 [bus notice] <cc
2018-12-15 22:31:49.218 [bus notice] <b8
2018-12-15 22:31:49.224 [bus notice] <c8
2018-12-15 22:31:49.229 [bus notice] <48
2018-12-15 22:31:49.234 [bus notice] <88
2018-12-15 22:31:49.239 [bus notice] <f2
2018-12-15 22:31:49.245 [bus notice] <a0
2018-12-15 22:31:49.248 [bus notice] <e0
2018-12-15 22:31:49.344 [bus notice] <5a
2018-12-15 22:31:49.349 [bus notice] <4e
2018-12-15 22:31:49.354 [bus notice] <50
2018-12-15 22:31:49.360 [bus notice] <0b
2018-12-15 22:31:49.364 [bus notice] <44
2018-12-15 22:31:49.369 [bus notice] <ce
2018-12-15 22:31:49.374 [bus notice] <5a
2018-12-15 22:31:49.379 [bus notice] <c4
2018-12-15 22:31:49.385 [bus notice] <c4
2018-12-15 22:31:49.391 [bus notice] <d7
2018-12-15 22:31:49.396 [bus notice] <e7
2018-12-15 22:31:49.489 [bus notice] <f0
2018-12-15 22:31:49.494 [bus notice] <c0
2018-12-15 22:31:49.499 [bus notice] <03
2018-12-15 22:31:49.504 [bus notice] <02
2018-12-15 22:31:49.509 [bus notice] <89
2018-12-15 22:31:49.513 [bus notice] <21
2018-12-15 22:31:49.518 [bus notice] <4a
2018-12-15 22:31:49.523 [bus notice] <b9
2018-12-15 22:31:49.527 [bus notice] <15
2018-12-15 22:31:49.534 [bus notice] <95
2018-12-15 22:31:49.539 [bus notice] <02
.......
`

When I start with scanconfig, I don't see any scan being done successfully. There are also those "max. symbols per second notices, that's quite strange. 
`pi@DomoPy:~ $ sudo ebusd -f --scanconfig=full --httpport=8787 --device=/dev/ttyUSB0
2018-12-15 22:24:01.536 [main notice] ebusd 3.2.v3.2 started with full scan
2018-12-15 22:24:01.768 [bus notice] bus started with own address 31/36
2018-12-15 22:24:01.791 [bus notice] signal acquired
2018-12-15 22:24:09.043 [bus notice] max. symbols per second: 1138
2018-12-15 22:24:10.036 [bus notice] max. symbols per second: 1273
2018-12-15 22:24:11.769 [main notice] starting initial full scan
2018-12-15 22:24:15.020 [bus notice] max. symbols per second: 1284
2018-12-15 22:24:16.041 [bus notice] max. symbols per second: 1289
2018-12-15 22:24:27.041 [bus notice] max. symbols per second: 1290
2018-12-15 22:24:35.040 [bus notice] max. symbols per second: 1300
2018-12-15 22:24:48.028 [bus notice] max. symbols per second: 1301
2018-12-15 22:26:11.827 [main notice] update check: revision v3.2-12-g45b9bad available, broadcast.csv: different version available
2018-12-15 22:26:12.010 [bus notice] max. symbols per second: 1304
2018-12-15 22:27:01.019 [bus notice] max. symbols per second: 1305
`

`pi@DomoPy:~ $ ebusctl i
version: ebusd 3.2.v3.2
update check: revision v3.2-12-g45b9bad available, broadcast.csv: different version available
signal: acquired
symbol rate: 1277
max symbol rate: 1304
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
`

The http interface gives the following JSON :
`
{
 "broadcast": {
  "messages": {
   "datetime": {
    "name": "datetime",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "error": {
    "name": "error",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "id-u": {
    "name": "id",
    "passive": true,
    "write": false,
    "lastup": 0
   },
   "signoflife": {
    "name": "signoflife",
    "passive": true,
    "write": false,
    "lastup": 0
   }
  }
 },
 "global": {
  "version": "3.2.v3.2",
  "updatecheck": "revision v3.2-12-g45b9bad available, broadcast.csv: different version available",
  "signal": true,
  "symbolrate": 1213,
  "maxsymbolrate": 1304,
  "qq": 49,
  "reconnects": 0,
  "masters": 1,
  "messages": 11,
  "lastup": 0
 }
}`

Any idea of what we missed?
Thanks in advance.
Rac'

john30, 2018-12-16:
are you sure no other tool is using the same device /dev/ttyUSB0? like fhem e.g.

john30, 2018-12-17:
check for messages being sent/received by using "ebusctl raw" for a start. then the ebusd.log should reveal whats on the wire. you could also go and check "ebusctl grab result"

Racailloux, 2019-01-05:
Hello back...

1st of all, happy new year!
I am back with the same problem, because of christmas break, this topic was delayed :)

Here is the result of grab, not really expressive! :)
> pi@DomoPy:~ $ ebusctl grab result
done

On the other hand, the raw data are really RAW :)
We checked quickly the data layer specs, but the starting of each raw data doesn't seem to look like a service request. And the bus notice are just incredibly huge.
Does the ebusd just don't split the request/response?
Does those data make sense to anybody (that are more expert on this protocol)?

If anything seems to be abnormal, that would help.
Thanks in advance.

> 2019-01-05 21:29:31.227 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-01-05 21:29:31.459 [bus notice] bus started with own address 31/36
2019-01-05 21:29:31.465 [bus notice] signal acquired
2019-01-05 21:30:11.966 [bus notice] <e0a0c1014399e1ca7df8f4001de1862ac9488c7d187499e9a0c1014b19e1ca7d7884041494fef02158f9c147871dd944951802001d4df3ddccc618c034224096c04c07890bc6a285a04a5881061d91845c20d90350fc8c1d4df3dd98e9c3f2a0047880841cb19190ab2dc4c440886a42f662200791006888955442e8c60050c34088c38a9a3b4b9a001960c9fe4c3b89d8c6a3088058e385a4fc6038321a90d10943d9e801ffe9a0c1014319e1ca7df8c91240403ec6760936e8c957c2d8c3c6c076f94b8cd2ca690185ed00fff80b054e72898bdd3fc21042ff074ce90d00a98bc90bca21f021b8404232790790d9b1ff0d2b8d48c7e308801b63e207ff601c1cd2d942408d40431d950804fa8c19e1065d98e9c3f2a069c9fac7a0c04c03e96590e894e86be03ec0db5830117809a9c880024b3dd98c4240808dc240f4b0214ab0ceb4c90459455fff4a88cc0e44a8b19987a208e030f02005f9c14687c84472590039806bc6402bcfca9190030050fc014b80424406dde580056049fe0b48c50f8ca8e9d194a09095c8b0c804a2cec44ab9d832c8010b014e72c18bdde5805532ffcb0c0620d983408942425d80038482884002c121d8415208586983f0d0509cc08064b0f18048a5e0888002c190d8415008c1614380b088434ace464bb0d8202a8239800050c34088c38a9a3b4b0ac8874199f8ae69834248a491e203ca80690f4310c9d04c8479fdc3c6c076f94b8cd2ca6941e6648148006905c686d0d0607084d881323ee9c844ceca8c8d9c001d4df3ddccc6183f34d8fc4e087a0110c2d0c27d48fc81021d9184b820d9035060800a81323ee9c846b6ca448d0cf18094f003c84ac84c11c10f49c8448040ffb84354804f8dd1d81c3d90c84958c84c2250c8415e400e1c8465cb2c0620d987408942425d015086fb80421cb10b908f2dc4c46b20fe0bc84c895f64f9f1a068c25bc9fe0dabb2428f0009594b49fc8040c0c84646094888c9e480101bca448410180b7ec82a20210cb2ff6be03ac0dbf34b49892a0c42c94800694519c1c9760f48642955e888019eccf84b184240804920fed1854c895f64f9f1a068499801fe0040c0c846440948880b8b809488f94acf48d945118c00c04f10e80d02e9d0c90f48a181a0f50132c94c78c18b5ded800868e8c64c07890bc6a285a04a581212c860380e1a90910943d9a8424240080490b18b48b08086a98154a845160bc88b6815fcc1b14042c3790790c844001bca4408101a9a3b4b089088e8fc012ec04e030030114b090d30f421b84042ca790790c88498509cc08064b06380c8e28b30e888018c3ab84b0a5478014cff8cf4a8045f986bc2fba06d0b800142c15208045d3bc04b1848e5ffb0214ab0cec44ab9155fff86a98176e845169dc88be090f0c98400694c19c1c9a14e3ce15b8bfe0042c74288c18a5d3bc823924958c84c2274dac25c900c0830c519ebc98400694c19c1cb094e1842f6f02158f0c170c8892bfe9528f06030fc4b58c50e8ca8e9d14af8c7a0014c03e965b0e8ca2101fe0042e74288c18a5d3bc8031ca4e591ff03c84a1a4c11c10f0929a038f6c1b140c0838fd00005838108c820fe0bc8c5475f64f9d150c8a8ff12a3408ca859864081b613ffe9e070804a194b48804a114042cac8874199f8ae698342482888c8494a8ccc0e44a831f08782b8433280468dd1d81c5274d1080490f18b48b0813c90f08194842005874089080713f02105f9c147871dd94495e183fe0710e80d00a9d0c90f4a61c8420d644243800c0490b18b48b0484bec8126144440ce2091b2421880c1b1c076f9428cd0ca91210df200c8096274dac25c901861248240f4f02005f9c14687c8c472598786fff02005f9c14687c84472d9a6028048ff4e88d00130c240c27dd203f38040c0c8464656d84154ff60380e5290d009435969019445824c07890fc6b485a04a580848fe12a3408ca859864081db82fa81323eb1017a8895a84260a8e221b8404232790790d9d1c94800694519c1c9764e2a8200ffe9e0b8c049c24b4a804a4680fa4e18844194c249c27dffc1b1c02d79429cd0ca91ff4e18840194c2d0c27d801da4fe81d80199e8ae20830350204982ff001bca440810180b7ec80385ec6403c84a1a4c11c10f0909f2702005f9c14687c84472d98619e9601c1cd2d942408d40427d058019ff6981cb004a104b48884803214ac7a0c04c03e965b0e8ca61b0a400fd88019c3ab84b1a5478014c84f3810f95d099a03c9c32424a1010b0fe80841cb19190ab2dc4c4ff6bc6402bcf4a91d0030044e591ff03c84ac84c11c10f09c800f0c080024b3dd98c4240808d103bf8815f0eb10183ce5a8cc4304bc299a0e70091b24210e8e00ac8874199f8ae69834248c00830e800c8e1862ac9418c7d1841045c5e0142c05208045dec8083123be8f02158f8c170c8882bfe95f02158b0c170c88929fe9579a17ef9c880024b3dd98c4240804d4c1c4281948420058740c908070480e3ff80420df90bd08f2d41540352d8fce7a0c04c03e96590e8946884c048ffb84354804f8dd1d81cb839015ac6402befca91900300508a40656038321a90d10943d9e8825091e3a0c18a884399f18b05f81068c9fa4e887a0110c2d0c27d2228748cf4a80405986bc2fba0a848c0ffc8874199f8ae69834248a02932ff815f0eb10187cec88cc461cf46fea0a0c1014399f1ca0ff80424e8c60bc8c50f8ca8e9d194284b8bff010b80424406dde5800c14a4fe5ac6402bb84085a80050f0cca4fc600c1c52d9d0400859e940e5995274d1080490f18b48b0018080f34e18844194c269c240e08604fa00c8e1860cc9418c7d188909b2f581260d5652d120d10352d808e1b4fef02158b0c170c8892bfe952538428148006905c686d0d040a096c0e9e070804a194b4880482024d8073aa142a9cbc9caca850eca03ff4a884c224288c4990e025482ff88019eccf84b18424080c9023842814800690519c0c9764ef081320eb10187cec88cc5f1438081323ee9c844cec88c8d44438019ff6981ca004a194b4880488000ff4d3b89c8c6e308801b63fd60380e5290d009435969b2a0c18a460cc9493b31038b40c8f680420df10b908f2dc444a239004a4e88d00130c2d0c27da9a9c70132c94c78c18b5dec80f9c1d80199e8ae60830390fe0dab8d488fe30ad41b63c38aa4fe81d80199e8ae2083035080c20200c8458114 [... cut ...] ce7a0c04c03e96590e8ca11fe04a9a101e045149dc88bc19401ff622005f9c946cf4a448d5495e58bfe0132c84c78c18b5dec804a2848ffb843
2019-01-05 21:31:12.624 [bus notice] <80468dd1f34e8cf80b484ea0898fdce5c298c1b1c0c0838fd0000583e10bc8c5474076f9d1500c90f060380e5a90d00943d9e901904bf60042c34288c18a5d3bc86380f34e18844194c249c27de2fa00c881147dd25c7d18e92d94f0814a006905c686d0d0a0b295c8b0880422cec64ab9d832586ccbfe4a88cc0e44a8b199d860b049fa0dab8d488fe30ad41b6342e81dffcb9484200587408908071030e5ff5ac6402bb84085a80050ffb843

Racailloux, 2019-01-06:
We did a fresh install of a raspbian (stetch this time instead of jessie) and only setup wifi + update + ebusd package.
When we plug the board on USB, the ttyUSB0 appears and the serial liked to it (so that looks totally logical)

We have quite the same result :
> pi@raspberrypi:~ $ ebusctl i
version: ebusd 3.3.v3.3
signal: acquired
symbol rate: 80
max symbol rate: 92
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

And the logs looks like this :
> 2019-01-06 13:26:04.713 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-01-06 13:26:05.065 [bus notice] bus started with own address 31/36
2019-01-06 13:26:05.094 [bus notice] signal acquired
2019-01-06 13:27:10.290 [bus notice] <f02158f8c14687c8448d0448e9a0b8c049c24b49892a80e4ff9488f94acf464ab9845f55f3a030f421b84f1a9491f843d969c049ff0142c34288c71a0b410ac5a80179ef8c40a8c190d041090869a142c04c03c9$
2019-01-06 13:27:20.732 [bus notice] <804636d3fbfe83008130fc58c809630cc9498cc8588691e3e981cb00d8990ac65a6200812ca8f30bc8c50f8ca8e999871224426be03e408f0009584b48680084f3d185ba478ca8e9d1daf019eba0010a884f19d0$
2019-01-06 13:27:27.424 [bus notice] <804636d3fbfec34d3b89d8c6b485a083836058ef00c8e1862ad94888f2a041a0210d03c84a1a4c118f3dd1c288c18741994069c2425864480019a1045f98414208c1004915b2fff80b014ee8c18b5decc00334fc$
2019-01-06 13:27:57.200 [bus notice] <804636d3fbfec362200e910187ce5ac4c4e0e0f8c180434b832fd00005830cd0c88740d99491d00300443c65ffc98400f94c11c10f49c8f86b60b2428f0094584b48425090ff8cf4a8c125d8415008e9410a15fc$
2019-01-06 13:28:08.841 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
2019-01-06 13:28:15.841 [bus notice] <804636d3fbfe43dce9cb98842005c70cc9760bfec78050404865b04380e8e8880d66e9a0b8c049c24b48892a0865ff81500eb191908f2d41544e1884409065b0e8d8e8f0ff424380080490f18b48f8c140ff0143$
2019-01-06 13:28:37.632 [bus notice] <fe074ce80d02e9c2d0c27d82cf47fee9a0b8c049c24b49892a204142e862200f91017a9591a880438a0cfc600c1cc2d942408942427a144bf6c88740d9569590030050008374c1c04343838fd20004830280f89c$
2019-01-06 13:28:49.828 [bus notice] <804636d3f34c2300429cf02005f9c146ceca448d204190f8e981cb00d8990ac65a6af08c40a8c190d0415008e9922542cb98842005870cc9760b0a58c8ff4e1884409465b0e8d8514710e80d02e9c26942401538$
2019-01-06 13:28:54.161 [bus notice] <8154e845149dc8cbc18c40a8040598415208c161008840a8c190d041090869218dc8a8ff98509cc080faf9f15080d8fc0bc8c50e8ca836e08792fe0d3b89d8c6a285a0ca58e2c1b14042ca698fd0d9f518c8fac7$
2019-01-06 13:29:01.338 [bus notice] <4bf481021d91014069834258f804e90b4c02a98bc90f4a61d020fec78050404865b0438068001bc38dc8c70091b2425018e5ff9488f94a09c64ab9845fc88840a8c190d0410908696c49fa0d3b89d8c6a285a04a

john30, 2019-01-06:
as said before, the data is just no ebus traffic but something else. check your system for any other process using the device e.g. using lsof and searching for ttyUSB0 in the output

closed 2019-01-06 by john30

Racailloux, 2019-01-06:
We red what you said before and that's the reason why we re-installed a fresh raspbian to avoid any modification that could influence this.
We rechecked it, and when ebusd service is stopped, no service is accessing to the ttyUSB0 (that is well connected).
When we start ebusd service, there are the following that are accessing to it :
> pi@raspberrypi:~ $ sudo lsof | grep ttyUSB0
ebusd     5348                  root    1uW     CHR      188,0      0t0      18259 /dev/ttyUSB0
bushandle 5348 5352             root    1uW     CHR      188,0      0t0      18259 /dev/ttyUSB0
network   5348 5353             root    1uW     CHR      188,0      0t0      18259 /dev/ttyUSB0
mainloop  5348 5354             root    1uW     CHR      188,0      0t0      18259 /dev/ttyUSB0

When we disconnect the ebus wires from the board, nothing appears in "raw logging mode".
When we reconnect it, the previously given data appears.

Does this means that the boards is doing a mess with the boiler data?
How could we test it?

Let me know where we should discuss about this? (as we are not german speaking, the fhem forums are quite weird :) )
Thanks for your help.

john30, 2019-01-06:
the thing is: the data of the ebusd.log is just not ebus traffic. this is something completely different.
so either you didn't connect to the right ppins on your heater, or the interface is broken, or another process is using the device. i have no other explanation

Racailloux, 2019-01-12:
As you expected, something else was writing on the USB-serial BUS : The hardware on which we connected.

The issue was coming from us, we connected the R.Pi to an Wemos D1 installed on the eBUS adapter v2.2. Actually, by default those hardware are setup to be used with a serial connection to a terminal (Putty). So the bytes we got in our logs are probably the menus displayed on this interface.

John30 gave me a track to use this interface :
> There is a setting for the ebusd-esp firmware, that allows using it as USB-serial bridge. Since the USB side occupies the RX/TX pins anyway, the connection to the adapter then has to be changed to D1 for the RX wirte and D4 for the TX wire. The settings needed on the ebusd-esp are these:
- eBUS RX+TX PINs: mixed software D1 + TX1 D4
- ebusd connection: port 0
it is still experimental though.

We actually used the UART provided separately with the hardware and I soldered it to be connected directly on the base board.
The boiler was not damaged (hopefully) and we were able to run the ebus and grab the data.

Thanks for the support, I hope that could help others to avoid to do the same error. (RTFM :) )