john30/ebusd #234: Explicit read from field1 not possible

rellla, 2018-12-18:
Hi,
i have an issue if i want to read only the first field:
```
root@smartmeter:/home/pi# ebusctl r -f -c ww_param2 temp_soll
05-051;TemperaturSoll;°C;50.0;10.0;48.0

root@smartmeter:/home/pi# ebusctl r -f -c ww_param2 temp_soll temp
48.0

root@smartmeter:/home/pi# ebusctl r -f -c ww_param2 temp_soll parameters
ERR: element not found in decode
```
Second field (temp) works, first field (parameters) fails.
Is this an issue or a feature?
Regards
rellla

john30, 2018-12-18:
depends on the final definition, so please post that as well by issuing "ebusctl f -f -c ww_param2 temp_soll"

rellla, 2018-12-18:
So here it is:
```
root@smartmeter:/home/pi# ebusctl f -f -c ww_param2 temp_soll
r,ww_param2,temp_soll,Normal Warmwassertemperatur 05-051,,15,0621,6592000e,command,s,TEM_P,0,,ParameterTemp10,type,s,UCH,0=none;2=on-off;4=Zeit;9=selection;13=Temperatur;64=Dauer;77=TemperaturSchwelle;141=TemperaturSoll;192=range,,Datentyp,unit,s,UCH,0=none;2=°C;4=K;8=%;10=h;12=min;16=kW;42=hh:mm,,Einheit,max_temp10,s,SIN,10,°C,Maximum,min_temp10,s,SIN,10,°C,Minimum,temp,s,SIN,10,°C,Mit dieser Einstellung wird der Warmwasser Sollwert eingestellt
```
I got it. I can ask for command, type, unit, max_temp10, min_temp10.

Thanks! Can be closed then :)

rellla

john30, 2018-12-18:
right

closed 2018-12-18 by john30