john30/ebusd #235: How do I spot invalid numeric argument in --def

semenoh, 2018-12-21:
I'm trying to define message like this:
```
pi@raspberrypi:~ $ ebusctl f -f | grep SetMode
uw,bai,SetMode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,

pi@raspberrypi:~ $ ebusctl w -def 'w,bai,wmode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,'
ERR: invalid numeric argument
```
So how can I figure out which one of there arguments is invalid ?

semenoh, 2018-12-21:
Ok, minor success:
By setting value it works:
```
pi@raspberrypi:~ $ ebusctl w -def 'w,bai,wmode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,' 0\;90.0\;-\;-\;0\;0\;0\;0\;0\;0
done
```

Although manual says value is optional:
```
pi@raspberrypi:~ $ ebusctl w ?
usage: write [-s QQ] [-d ZZ] -c CIRCUIT NAME [VALUE[;VALUE]*]
  or:  write [-s QQ] [-d ZZ] -def DEFINITION [VALUE[;VALUE]*]
```
But still I can't use it for some reason:
```
pi@raspberrypi:~ $ ebusctl w -c bai wmode 0\;70.0\;-\;-\;0\;0\;0\;0\;0\;0
ERR: element not found
```
And `ebusctl f | grep wmode` does not find anything.

semenoh, 2018-12-21:
BTW After running this command 
```
ebusctl w -def 'w,bai,wmode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,' 0\;90.0\;-\;-\;0\;0\;0\;0\;0\;0
```
`SetMode` and `FlowTempDesired` are set but boiler stops heating (so `FlowTemp` and `ReturnTemp` decrease)
[image]

andig, 2018-12-21:
@semenoh not sure what your graph really shows but- if you send all zeros- I'm convinced you may have a set a temperature but also turned heating off. So not wonder temp goes down?

semenoh, 2018-12-22:
> @semenoh not sure what your graph really shows but- if you send all zeros- I'm convinced you may have a set a temperature but also turned heating off. So not wonder temp goes down?

Graph shows FlowTempDesired (blue), FlowTemp (pink) and ReturnTemp (yellow). At the very end of the graph FlowTempDesired goes up, but FlowTemp and Return temp goes down. 
I've been thinking about zeros. When I set temperature with VRT 50 thermostat I can see message with zeros. That's why I use zeros. But running `ebusctl r -V -c bai SetMode` looks like first zero after dashes is `disablehc`. So I tried sending ` 0\;70\;-\;-\;1\;0\;0\;0\;0\;0`. But it looks like it has no effect:
```
$ ebusctl w -def 'w,bai,wmode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,' 0\;70\;-\;-\;1\;0\;0\;0\;0\;0
done

$ ebusctl r -c bai SetMode
auto;70.0;-;-;0;0;0;0;0;0
```
But that's for another issue.

Main reason for this issue is that I expect to be able to use command I defined without definition. Like this:

```
$ ebusctl w -def 'w,bai,wmode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,' 0\;90.0\;-\;-\;1\;0\;0\;0\;0\;0
done

$ ebusctl w wmode 0\;60.0\;-\;-\;1\;0\;0\;0\;0\;0
```
Or is it supposed to be used with definition every time ?

@andig could you please show an example how you defined your `SetModeOverride` command and how you use it ?

john30, 2018-12-22:
the docu nowehere states that the definition passed after "-def" is stored for future requests