john30/ebusd #236: Wolf CGW-2 compatibility

WoMec, 2018-12-21:
Hi,  I have following setup: 

 - ebusd running on Rpi Zero W
 - ebus adapted v2.2
 - Wolf CGW-2 (boiler + DWH)

... and I probably have problem with configuration. I can see messages on ebus, I can see devices on ebus, but I'm unable to translate raw messages to the human readable form. 

```
pi@ebusd:~ $ service ebusd status
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/systemd/system/ebusd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-12-21 17:57:07 CET; 6min ago
  Process: 6769 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=0/SUCCESS)
 Main PID: 6770 (ebusd)
   CGroup: /system.slice/ebusd.service
           └─6770 /usr/bin/ebusd --scanconfig -d /dev/ttyAMA0 --latency=20000 --httpport=8080

Dec 21 17:57:07 ebusd systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
Dec 21 17:57:07 ebusd systemd[1]: ebusd.service: PID file /var/run/ebusd.pid not readable (yet?) after start: No
Dec 21 17:57:07 ebusd systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..

```

```
pi@ebusd:~ $ ebusctl i
version: ebusd 3.2.v3.2
update check: revision v3.2-12-g45b9bad available, broadcast.csv: different version available
signal: acquired
symbol rate: 12
max symbol rate: 167
min arbitration micros: 53
max arbitration micros: 88
min symbol latency: 17
max symbol latency: 17
reconnects: 0
masters: 5
messages: 16
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Wolf;ID= !;SW=8000;HW=0000"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3
address 36: slave #8, ebusd
address 50: slave
address f1: master #10
address ff: master #25
```

```
pi@ebusd:~ $ ebusctl grab result
30fe03010e0000008000000000000000000000 = 10
30fe50230528000a1400 = 4
30fe5023054423000000 = 4
30fe502305844e2a0000 = 4
f1fe050308010100ff51ff2600 = 9
f1fe0800080005190000a0002d = 10
03fe050308010048ff541c2600 = 28
3003502203cc0100 = 4
3008502203cc0c00 / 020000 = 5
3008502203cc0d00 / 02bb01 = 7
3008502203cc0e00 / 028001 = 4
3008502203cc1227 / 020000 = 5
3008502203cc1327 / 020000 = 5
3008502203cc1600 / 021e01 = 7
3008502203cc1a27 / 029800 = 6
3008502203cc2602 / 025e51 = 7
3008502203cc2702 / 020000 = 7
3008502203cc2802 / 028f26 = 7
3008502203cc2902 / 020000 = 6
3008502203cc2a02 / 02980a = 6
3008502203cc2d00 / 029600 = 4
3008502203cc3127 / 020000 = 5
3008502203cc3227 / 020000 = 5
3008502203cc342a / 020000 = 5
3008502203cc352a / 020000 = 5
3008502203cc3727 / 020100 = 6
3008502203cc5327 / 020a00 = 6
3008502203cc5427 / 020800 = 6
3008502203cc5927 / 020000 = 5
3008502203cc5a27 / 020000 = 5
3008502203cc6327 / 02dc05 = 6
3008502203cc6427 / 020200 = 5
3008502203cc6a27 / 020080 = 6
3008502203cc6c01 / 020000 = 7
3008502203cc6f01 / 021900 = 7
3008502203cc7027 / 020000 = 5
3008502203cc7301 / 025600 = 7
3008502203cc8813 / 020000 = 5
3008502203ccbf0a / 020200 = 5
30ff502203cca82a = 4
30ff502203cca92a = 4
ff08502203117301 / 025600 = 9
ff08502203241600 / 021e01 = 5
ff085022033a2902 / 020000 = 5
ff085022034d7201 / 020800 = 4
ff085022035c0100 / 020000 = 5
ff08502203662802 / 028f26 = 5
ff08502203771a27 / 028a00 = 5
ff08502203aa2602 / 025e51 = 5
ff08502203cc0e00 / 028001 = 5
ff08502203f62702 / 020000 = 4
ff355022031f8727 / 029601 = 5
ff3550220327ae27 / 020200 = 5
ff355022032bb527 / 02ecff = 5
ff355022034bc327 / 020100 = 5
ff35502203541300 / 02c201 = 5
ff3550220357ab27 / 020700 = 5
ff355022035c0100 / 020000 = 5
ff3550220377b427 / 020080 = 5
ff355022037baf27 / 020200 = 5
ff355022037fa627 / 020000 = 5
ff355022039fac27 / 020100 = 5
ff35502203a37527 / 020100 = 5
ff35502203c3ad27 / 020200 = 5
ff35502203f39627 / 020600 = 5
ff35502203fbd327 / 020000 = 5
ff35502203ff7427 / 020100 = 5
3003050709bb048b020080ff5a00 = 22
305050140703b3280019005a / 09000100800080008828 = 23
03f10800080080000080020005 = 9
ff0850220b54532754270c0002000300 / 0a0a00080000009601c201 = 5
ff0850220b550d000e0016006c016f01 / 0a970180011e0100001900 = 5
ff0850220b8b532702000d006f015727 / 0a0a009601b10118001b00 = 5
ff0850220b852a0219271a2757274801 / 0a980a6b0c980037002100 = 5
ff3550220b1375271300c327a727a827 / 0a0100c201010000157425 = 5
ff3550220b17a12785278114b5277427 / 0a000000001500ecff0100 = 5
ff3550220b0ba927aa27b127b027a227 / 0a00127425011274250015 = 5
ff3550220b03a327a427a527b127b027 / 0a74250012742501127425 = 5
```

```
pi@ebusd:~ $ tail -30 /var/log/ebusd.log
2018-12-21 18:02:47.746 [update notice] received unknown MS cmd: ff08502203117301 / 021600
2018-12-21 18:02:47.763 [bus notice] <ff08502203117301fd000216007200
2018-12-21 18:02:47.970 [update notice] received unknown MS cmd: 305050140703b3280019005a / 09000100800080008828
2018-12-21 18:02:47.971 [bus notice] <305050140703b3280019005a6b00090001008000800088284900
2018-12-21 18:02:48.036 [update notice] received unknown MS cmd: ff35502203fbd327 / 020000
2018-12-21 18:02:48.037 [bus notice] <ff35502203fbd327d2000200002c00
2018-12-21 18:02:48.327 [update notice] received unknown MS cmd: ff0850220b8b532702000d006f015727 / 0a0a009601eb0119001900
2018-12-21 18:02:48.344 [bus notice] <ff0850220b8b532702000d006f01572774000a0a009601eb01190019000900
2018-12-21 18:02:48.568 [update notice] received unknown MS cmd: ff08502203771a27 / 028800
2018-12-21 18:02:48.585 [bus notice] <ff08502203771a2736000288003e00
2018-12-21 18:02:48.810 [update notice] received unknown MS cmd: ff08502203241600 / 021f01
2018-12-21 18:02:48.827 [bus notice] <ff08502203241600d200021f01f100
2018-12-21 18:02:49.051 [update notice] received unknown MS cmd: ff08502203cc0e00 / 028101
2018-12-21 18:02:49.068 [bus notice] <ff08502203cc0e00d100028101bd00
2018-12-21 18:02:49.928 [update notice] received unknown BC cmd: 30fe03010e0000008000000000000000000000
2018-12-21 18:02:49.929 [bus notice] <30fe03010e00000080000000000000000000007d
2018-12-21 18:02:51.006 [update notice] received unknown BC cmd: f1fe0800080005190000a0002d
2018-12-21 18:02:51.006 [bus notice] <f1fe0800080005190000a0002d4a
2018-12-21 18:02:55.418 [update notice] received unknown BC cmd: 03fe050308010048ff621c2600
2018-12-21 18:02:55.418 [bus notice] <03fe050308010048ff621c2600d4
2018-12-21 18:02:56.337 [update notice] received unknown BC cmd: f1fe050308010100ff62ff2600
2018-12-21 18:02:56.337 [bus notice] <f1fe050308010100ff62ff2600e9
2018-12-21 18:02:58.399 [update notice] received unknown MM cmd: 3003050709bb048b020080ff5a00
2018-12-21 18:02:58.399 [bus notice] <3003050709bb048b020080ff5a00a300
2018-12-21 18:02:58.457 [update notice] received unknown MM cmd: 03f10800080080000080020005
2018-12-21 18:02:58.457 [bus notice] <03f108000800800000800200059500
2018-12-21 18:03:00.533 [update notice] received unknown MS cmd: 305050140703b3280019005a / 09000100800080008828
2018-12-21 18:03:00.541 [bus notice] <305050140703b3280019005a6b00090001008000800088284900
2018-12-21 18:03:02.888 [update notice] received update-read broadcast datetime QQ=30: -;18:03:10;21.12.2018
2018-12-21 18:03:02.907 [bus notice] <30fe07000900801003182112051863
```

john30, 2018-12-22:
indeed. this is a configuration issue rather than an ebusd one, but anyway, you could give the https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de a try by downloading it (or cloning the repo) and pointing ebusd to the ebusd-2.x.x/de folder using the ["-c" option](https://github.com/john30/ebusd/wiki/2.-Run#message-configuration-options) in /etc/default/ebusd

closed 2018-12-22 by john30

WoMec, 2018-12-22:
And are the wolf config files included in online configuration? (http://ebusd.eu/config/)

john30, 2018-12-23:
no, as the online config only covers what is behind the ["latest" link](https://github.com/john30/ebusd-configuration/blob/master/latest)

WoMec, 2018-12-23:
I tried to run in with following params. 

`EBUSD_OPTS="--scanconfig -c /home/pi/ebusd-configuration/ebusd-2.x.x/de -d /dev/ttyAMA0 --latency=20000 --httpport=8080 --lograwdatasize=10000"
`

result is not so good ... 

```
pi@ebusd:~ $ ebusctl i
version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 50
max symbol rate: 156
min arbitration micros: 57
max arbitration micros: 63
min symbol latency: 17
max symbol latency: 17
reconnects: 0
masters: 5
messages: 16
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Wolf;ID= !;SW=8000;HW=0000"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Wolf;ID=;SW=0204;HW=0100"
address 36: slave #8, ebusd
address 50: slave
address f1: master #10
address ff: master #25
```

```
pi@ebusd:~ $ ebusctl grab result
30fe03010e4c00008000000000000000000000 = 5
30fe50230528000a1400 = 2
30fe5023054423000000 = 2
30fe502305844e2a0000 = 2
f1fe050308010100ff5cff2b05 = 5
f1fe0800080005330500a0002d = 5
03fe050308010048ff5c1e2b07 = 15
3003502203cc0100 = 3
3008502203cc0200 / 029301 = 1
3008502203cc0c00 / 024c00 = 2
3008502203cc0d00 / 02cd01 = 3
3008502203cc0e00 / 02b001 = 3
3008502203cc1227 / 020000 = 2
3008502203cc1327 / 020000 = 2
3008502203cc1600 / 022f01 = 7
3008502203cc1627 / 0240fe = 1
3008502203cc1727 / 020080 = 1
3008502203cc1a27 / 028c00 = 3
3008502203cc1d27 / 025e01 = 1
3008502203cc2602 / 02d651 = 3
3008502203cc2702 / 020000 = 3
3008502203cc2802 / 02bb26 = 3
3008502203cc2902 / 020000 = 3
3008502203cc2a02 / 02a90a = 3
3008502203cc2d00 / 029600 = 3
3008502203cc3127 / 020000 = 2
3008502203cc3227 / 020000 = 2
3008502203cc342a / 020000 = 3
3008502203cc352a / 020000 = 3
3008502203cc362a / 02c800 = 1
3008502203cc3727 / 020100 = 3
3008502203cc372a / 02c800 = 1
3008502203cc4f27 / 024600 = 1
3008502203cc5327 / 020a00 = 3
3008502203cc5427 / 020500 = 3
3008502203cc5727 / 021e00 = 1
3008502203cc5927 / 020000 = 3
3008502203cc5a27 / 020000 = 3
3008502203cc5e27 / 025c00 = 1
3008502203cc6327 / 02dc05 = 3
3008502203cc6427 / 020200 = 3
3008502203cc6a27 / 020080 = 3
3008502203cc6c01 / 020000 = 7
3008502203cc6f01 / 021900 = 3
3008502203cc7027 / 020000 = 3
3008502203cc7301 / 021600 = 3
3008502203cc8813 / 020000 = 2
3008502203ccbf0a / 020200 = 2
30ff502203cca82a = 2
30ff502203cca92a = 2
ff08502203117301 / 021600 = 4
ff08502203241600 / 022e01 = 2
ff085022033a2902 / 020000 = 2
ff085022034d7201 / 020800 = 2
ff085022035c0100 / 020000 = 3
ff08502203662802 / 02bb26 = 3
ff08502203771a27 / 028c00 = 2
ff08502203aa2602 / 02d651 = 2
ff08502203cc0e00 / 02b301 = 2
ff08502203f62702 / 020000 = 2
ff355022031f8727 / 02ce01 = 3
ff3550220327ae27 / 020200 = 3
ff355022032bb527 / 020000 = 3
ff355022034bc327 / 023400 = 3
ff35502203541300 / 02c201 = 3
ff3550220357ab27 / 020700 = 3
ff355022035c0100 / 020000 = 3
ff3550220377b427 / 020080 = 3
ff355022037baf27 / 020200 = 3
ff355022037fa627 / 020000 = 3
ff355022039fac27 / 020100 = 3
ff35502203a37527 / 020100 = 3
ff35502203c3ad27 / 020200 = 3
ff35502203f39627 / 020600 = 3
ff35502203fbd327 / 024c00 = 2
ff35502203ff7427 / 020100 = 3
3003050709bb0486020080ff5a00 = 12
30505014070366280019005a / 09000100800080008828 = 12
03f10800080080990780020005 = 5
ff0850220b54532754270c0002000300 / 0a0a0005004c009301c201 = 3
ff0850220b550d000e0016006c016f01 / 0acd01b2012f0100001900 = 3
ff0850220b8b532702000d006f015727 / 0a0a009301cc0119001e00 = 2
ff0850220b852a0219271a2757274801 / 0aa90a090e8c001e002100 = 3
ff3550220b1375271300c327a727a827 / 0a0100c201340027107625 = 3
ff3550220b17a12785278114b5277427 / 0a00000000160000000100 = 3
ff3550220b0ba927aa27b127b027a227 / 0a270d7625270d76252710 = 3
ff3550220b03a327a427a527b127b027 / 0a7625270d7625270d7625 = 3
```

john30, 2018-12-25:
ok, please check the log file for errors especially with regard to loading a config file

WoMec, 2018-12-25:
nothing special there regarding to problems with config file

```
2018-12-25 13:45:48.941 [main notice] ebusd 3.2.v3.2 started with auto scan
2018-12-25 13:45:49.026 [bus notice] bus started with own address 31/36
2018-12-25 13:45:49.081 [bus notice] signal acquired
2018-12-25 13:45:49.599 [bus notice] new master 30, master count 2
2018-12-25 13:45:49.600 [update notice] received unknown BC cmd: 30fe03010e2300008000000000000000000000
2018-12-25 13:45:52.637 [bus notice] new master 03, master count 3
2018-12-25 13:45:52.638 [update notice] received unknown MS cmd: 3008502203cc0c00 / 022300
2018-12-25 13:45:52.791 [update notice] received unknown BC cmd: 03fe050308010040ff3b1d2b03
2018-12-25 13:45:52.943 [update notice] received unknown MS cmd: 3008502203ccbf0a / 020200
2018-12-25 13:45:53.269 [update notice] received unknown MS cmd: 3008502203cc8813 / 020000
2018-12-25 13:45:53.681 [update notice] received unknown MS cmd: 3008502203cc1227 / 020000
2018-12-25 13:45:54.008 [update notice] received unknown MS cmd: 3008502203cc1327 / 020000
2018-12-25 13:45:54.334 [update notice] received unknown MS cmd: 3008502203cc3127 / 020000
2018-12-25 13:45:54.660 [update notice] received unknown MS cmd: 3008502203cc3227 / 020000
2018-12-25 13:45:54.814 [bus notice] new master f1, master count 4
2018-12-25 13:45:54.815 [update notice] received unknown MM cmd: 03f10800080080800380020005
2018-12-25 13:45:54.994 [update notice] received unknown MM cmd: 3003050709bb04ac020080ff5a00
2018-12-25 13:45:55.304 [update notice] received unknown MS cmd: 3008502203cc5927 / 020000
2018-12-25 13:45:55.461 [bus notice] new master ff, master count 5
2018-12-25 13:45:55.461 [update notice] received unknown MS cmd: ff085022035c0100 / 020000
2018-12-25 13:45:55.618 [update notice] received unknown MS cmd: 3008502203cc5a27 / 020000
2018-12-25 13:45:55.688 [update notice] received unknown MS cmd: ff355022035c0100 / 020000
2018-12-25 13:45:56.517 [main error] scan config 04: ERR: read timeout
2018-12-25 13:45:57.188 [update notice] received unknown MS cmd: 3008502203cc6427 / 020200
2018-12-25 13:45:57.430 [update notice] received unknown MS cmd: ff08502203117301 / 021600
2018-12-25 13:45:57.568 [update notice] received unknown MS cmd: 305050140703b32a0019005a / 09000100800080008828
2018-12-25 13:45:57.633 [update notice] received unknown MS cmd: ff35502203fbd327 / 022300
2018-12-25 13:45:57.838 [update notice] received unknown MS cmd: ff0850220b8b532702000d006f015727 / 0a0a00ab01290100001e00
2018-12-25 13:45:57.909 [update notice] received unknown MS cmd: 3008502203cc7027 / 020000
2018-12-25 13:45:58.029 [bus notice] max. symbols per second: 129
2018-12-25 13:45:58.066 [update notice] received unknown MS cmd: ff08502203771a27 / 028b00
2018-12-25 13:45:58.223 [update notice] received unknown MS cmd: 3008502203cc342a / 020000
2018-12-25 13:45:58.294 [update notice] received unknown MS cmd: ff08502203241600 / 022701
2018-12-25 13:45:58.450 [update notice] received unknown MS cmd: ff08502203cc0e00 / 02b501
2018-12-25 13:45:58.520 [update notice] received unknown BC cmd: f1fe0800080005e60100a0002d
2018-12-25 13:45:58.691 [main error] scan config 08: ERR: read timeout
2018-12-25 13:45:58.850 [update notice] received unknown MS cmd: 3008502203cc352a / 020000
2018-12-25 13:45:59.944 [update notice] received unknown MS cmd: 3008502203cc0d00 / 022901
2018-12-25 13:46:00.271 [update notice] received unknown MS cmd: 3008502203cc1600 / 022801
2018-12-25 13:46:00.598 [update notice] received unknown MS cmd: 3008502203cc6c01 / 020000

```

john30, 2018-12-25:
> 2018-12-25 13:45:58.691 [main error] scan config 08: ERR: read timeout

this is the problem. you can try increasing the slave answer timeout

WoMec, 2018-12-25:
I raised latency of slave up to 200ms but with no luck. I'm still getting read timeout ... should I try to install **ttyebus** module instead?

WoMec, 2018-12-25:
I probably get it working ... ttyebus module works fine :)

```
pi@ebusd:~/ttyebus $ ebusctl i
version: ebusd 3.2.v3.2
update check: revision v3.2-25-gc475e6c available, broadcast.csv: different version availabl
signal: acquired
symbol rate: 88
max symbol rate: 158
min arbitration micros: 37
max arbitration micros: 69
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 5
messages: 61
conditional: 0
poll: 0
update: 6
address 03: master #11
address 08: slave #11, scanned "MF=Wolf;ID= !;SW=8000;HW=0000", loaded "wolf/08.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Wolf;ID=;SW=0204;HW=0100"
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Wolf;ID= !;SW=8000;HW=0000", loaded "wolf/50.csv"
address f1: master #10
address ff: master #25

```

john30, 2018-12-26:
> OK, I'm getting some information, but it's strange that all HG parameters are empty.

all the data is not automatically polled by ebusd, so each message shown as "no data stored" was just not intercepted by ebusd so far. you'd have to actively query those e.g. by issuing "ebusctl read hg90".

> Also I would give it a try a translate config files to en.

appreciated :)

> Do you have some co-op tool for translations?

unfortunately not. I'm still trying to get configs in version 3 ready, which allow e.g. having several translations in one single CSV file. But my time just does not allow it...

>And is there some documentation to update config for device on address 35?

basically it is try and error, but in principle it is [described here](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)

WoMec, 2018-12-28:
I found openhab2 binding for Wolf boiler ... is it posible to somehow transform those configs for ebusd? 

[https://github.com/csowada/openhab-bindings/blob/41886c8658c0a9c3f1dcf0872fc3d3f458d8ff56/org.openhab.binding.ebus/src/META-INF/wolf-configuration.json](https://github.com/csowada/openhab-bindings/blob/41886c8658c0a9c3f1dcf0872fc3d3f458d8ff56/org.openhab.binding.ebus/src/META-INF/wolf-configuration.json)