john30/ebusd #237: polling data results in HcPump start

sbrauns77, 2018-12-23:
HI,
Thanks for a great piece of software.
I am just starting using it and finding my way around in it.
I was polling data via MQTT every 10 seconds and noticed at night when the heater was set to a low value the internal pump starts very often for a very short time. Average it would start every 3 minutes. Last night it had start about 250 times.
With HcPumpStarts you can check the amount of times the pump started.
As soon as I stop polling the data the pump counter doesn't increase anymore.
Am i missing something? anybody else seeing this?

thanks,
Sander

sbrauns77, 2018-12-24:
I had ebusd running last night but I wasn't polling for any data, counter didn't increase anymore.
```
sander@raspberrypi5:~ $ ebusctl r -v HcPumpStarts
bai HcPumpStarts cntstarts2=3231

```

andig, 2018-12-24:
Imho if the BAI doesn‘t send it and you don‘t force an active read ebusd will always give you the cache value which doesn‘t change. Just use -f for an active read.

sbrauns77, 2018-12-24:
Thanks, did not know that. But still the pump didn't start in the last 24H when I was not polling for data.
```
sander@raspberrypi5:~ $ ebusctl r -f -v HcPumpStarts
bai HcPumpStarts cntstarts2=3231
```

sbrauns77, 2018-12-24:
I am pretty sure the  pump is starting and stopping. You can even see it in the live monitor that it runs through **S.02** : *Pump pre-run* and **S.07** : *Pump overrun* for just a second.
```
ebusctl read -f -v PumpHours
bai PumpHours hoursum2=188
```

sbrauns77, 2018-12-26:
Would it be possible for anybody that doesn't run his heating at night check the HcPumpStarts before going to sleep and next day. I just want to know if this is something only me is experiencing.
Thanks

andig, 2018-12-26:
Mine is running. Did you check the hours before and after?

sbrauns77, 2019-01-02:
I just wanted to let you know I found the issue I had. It had to do with an other master being active on the bus. I have an Vaillant ecotec plus, and I noticed that when I removed the thermostat from the ebus the heater just started up. Looking in the ebus log I saw that master 7f was commanding the heater to start. After some digging I found that inside my heater there was a opentherm module installed from factory. earlier models didn't come with this PCB installed as standard but you had to bye as an option. I just disconnected this PCB inside the heater and found that the heater didn’t start anymore without an thermostat installed. Also the command ebusctl info didn’t show the 7f master anymore. 
This ended up also being the one that was restarting the pump for some reason when data was polled.

Thanks for your support. Keep up the good work.
Thanks

john30, 2019-01-02:
interesting, good to know!

closed 2019-01-02 by john30

Dinth, 2019-01-02:
> With HcPumpStarts you can check the amount of times the pump started.

This may be a little bit out of the topic, but are there any interesting uses for data from HcPumpStarts? I thought it's purely for maintanence/gas engineer use?

john30, 2019-01-02:
indeed it is. anyway: it tells you how often the HC pump was started. that's it.