john30/ebusd #239: Docker Compose für Wemos Esp

chris8519, 2018-12-27:
Hallo wollte mal fragen wie die Docker compose für den Wemos aussehen müsste damit dieser direkt durchgereicht wird .. mfg

ebusd:
        restart: always
        image : john30/ebusd
        #expose:
        #    -"8888"
        #ports:
        #    -"8888:8888"
        #devices:
        # - "/dev/ttyACM0:/dev/ttyACM0"
        network_mode: host

john30, 2018-12-28:
äh für den Wemos? via USB oder wie? eigentlich doch WLAN, oder nicht? Dann müsstest Du halt die IP Adresse als "-d <ip>" Parameter an das run übergeben (samt --scanconfig und sonstigem)

chris8519, 2018-12-29:
Ja Sorry als netzwerkgerät ... 
kann man den Befehl : docker run --rm -it -p 8888 john30/ebusd -f --scanconfig -d udp:192.168.178.123:10000 --latency=80000

irgendwie in in eine configuration.yaml einbauen damit dieser direkt mit startet und nach dem netzwerkgerät sucht .

andig, 2018-12-29:
Sicher. Was genau versuchst Du zu erreichen? Wieso docker-compose?

chris8519, 2018-12-29:
Um so wenig wie möglich einstellungen vornehmen zu müssen damit das system nach einem ausfall oder neuinstallation direkt wieder läuft ...

john30, 2018-12-30:
du kannst auch einfach mit `docker run --restart=always` starten, dann macht der dockerd das beim System-Neustart automatisch

andig, 2019-11-04:
@chris8519 das Issue kann geschlossen werden?

domi-a, 2020-09-27:
> Sicher. Was genau versuchst Du zu erreichen? Wieso docker-compose?

weil docker-compose immer besser ist!
auch in diesem fall weil man seine config an einem einzigen ort hat.

aber zur sache - da auch ich lange danach gesucht habe - und es evtl jahre später immer noch gesucht wird:
die arguments kannst du mit "command" übergeben
bsp: command: ["-f", "--scanconfig","-d", "tcp:192.168.1.33:9999", "--latency=80000"]

closed 2020-10-03 by chris8519