john30/ebusd #240: Beginner problem with two Brink/Encon Renovent 300 units

mstroeve, 2018-12-28:
I am trying to read data from two Brink/Encon Renovent 300 units which are configured in a master/slave setup and are connected using the ebus connection on these units.

For the reading I use a Esera USB Ebus coupler which is also connected.
I found this mapping CSV https://github.com/dstrigl/ebusd-config-brink-renovent-excellent-300
Which I placed in /etc/ebusd/encon/3c.csv

When I issued the following command it seem to find the master and slave (37 and 56?).

`/etc/ebusd$ ebusctl i
version: ebusd 3.3.v3.3
update check: revision v3.3-4-g212b22d available
signal: acquired
symbol rate: 21
max symbol rate: 38
reconnects: 0
masters: 2
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
address 56: slave`

But when I try for example: 
`ebusctl read -f SoftwareVersion
ERR: element not found`

it doesn't seem to work.

The strange thing is that's after a while an extra slave comes up with "ebusctl i":
`address 3c: slave #18, scanned "MF=ENCON;ID=  ;SW=-;HW=-"`

which seems to be te address mentioned in the csv file. 

Any hints how I could get some read info from the master and slave unit?

john30, 2018-12-30:
how did you start the daemon?

john30, 2018-12-30:
hm, your arguments to ebusd don't match your previous comment where the 3c appears as scanned device. this is usually only the case if ebusd was started with "--scanconfig". are you sure that you have only one ebusd instance running? check by issuing `ps aux|grep ebusd`

mstroeve, 2018-12-30:
I checked my bash_history and I tried some other things I now have a reading. nice!
But how can i read/write info from the slave unit as it seems I can only get info from the master

I deleted all the csv files from the mentioned GitHub repo all except _7c.renovent-excellent-400.csv_ 

when i start the config seems ok:
`sudo ebusd --checkconfig -c /etc/ebusd/
2018-12-30 13:03:37.093 [main notice] ebusd 3.3.v3.3 performing configuration check...
2018-12-30 13:03:37.118 [main notice] found messages: 75 (0 conditional on 0 conditions, 0 poll, 0 update)
2018-12-30 13:03:37.119 [main notice] ebusd stopped`

and then I started Ebusd using:
`ebusd -c /etc/ebusd/ -d /dev/ttyUSB-Ebus`

when I open ebusctl in a different ssh terminal I can fire the command read Aussenlufttemperatur and I get 9.9 which is the temp also shown on the display of the master unit.

john30, 2018-12-30:
ok, so is there any issue left?

mstroeve, 2018-12-30:
well I am trying to read values from the slave and master unit but I always seem to get the same value while the displays state different values for example (Ablufttemperatur) 19.6 and 20.

I tried the following different addresses using it a both -s as -d override address.
But I always get 19.6 (or invalid address for some entries)
 
`version: ebusd 3.3.v3.3
update check: revision v3.3-4-g212b22d available
signal: acquired
symbol rate: 18
max symbol rate: 40
min arbitration micros: 714
max arbitration micros: 1200
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 2
messages: 76
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
address 3c: slave #18, scanned "MF=ENCON;ID=  ;SW=-;HW=-"
address 56: slave`

john30, 2019-01-01:
so the display shows 20 while reading from bus is 19.6? well this is just a simple rounding done by the display I guess.

closed 2019-01-01 by john30