john30/ebusd #242: Automatic execution of read after write

peti69, 2018-12-28:
Hi John,

Is there a way to automatically execute a read operation after a write operation has been finished? This would provide an immediate feedback on the outcome of the write operation to the initiator of the write operation and would inform all other subscribers (MQTT) about the new value. 

Kind regards,
Peter

john30, 2018-12-30:
no, but a write op should also trigger an MQTT topic being sent for the wrote message

peti69, 2019-01-05:
Ok. Thanks.

peti69, 2019-01-05:
ebusd publishes a MQTT message with topic ebus/write_msg/field in case it receives a MQTT message ebus/write_msg/field/set.  In case ebus messages with different field layouts are required for reading and writing the new value is published with a topic (ebus/write_msg/field) different to the one used for value polling (e.g. ebus/read_msg/field).

john30, 2019-01-06:
is that a question or what do you want to say exactly?

peti69, 2019-01-06:
It is a reaction to your first comment where you said that ebusd publishes a MQTT message for subscribers when performing a write (and not directly related to my original question). Depending on the messages used for reading and writing the topic for publishing the new value towards MQTT is not the same as the one used for value polling. That is, different MQTT topics are used for publishing the same data element. I think this is not an ideal situation and should/could be improved.

john30, 2019-01-06:
then your message configuration is bad. it is intended to have read and write messages for the same circuit and same content sharing the same message name. if so, ebusd will publish with the same topic as well

peti69, 2019-01-07:
I have a read message which returns two data elements. For each of these two data elements there is a dedicated write message transporting only one data element. Therefore three different MQTT topics.

john30, 2021-10-06:
not feasible for now and needed urgently as it seems, so closed for now. feel free to reopen if this is really needed