john30/ebusd #247: Using read -f with Vaillant VRT350 only works once to send read on the bus

congoblue, 2019-01-03:
Hi, thanks for your work with ebusd, it's a great resource. I am using ebusd on a Raspberry Pi with a Vaillant Ecofit Pure 625 and Vaillant VRT350, just to monitor the system operation. I have built the ebus adapter v2 and am using the Pi's PL011 uart with the ttyebus no-latency driver which is all working great.

I had to use the VRT370 config file to get the VRT350 working - seems to be a firmware version thing.

Nearly all the values I want to monitor are already on the bus, being sent between the boiler and the VRT350 in status messages. However I am also trying to read "DisplayedRoomTemp" from the VRT350. Using either TCP read -f command, or http maxage or required command, the log shows a single "read" command being sent on the bus, but after that one message, further calls to read -f do not cause a "read" message to be sent - there's nothing in the log at all. The console returns "done" but no actual read command is sent. If the command is sent without -f, the cached value is returned. If I stop ebusd and start it again, it will work one more time again.

Successful read looks like this:
2019-01-01 13:07:16.672 [update notice] sent read 350 DisplayedRoomTemp QQ=31: 20.56

I think I should be able to send the read -f command as often as needed to cause a read on the ebus for the latest value? Am I doing something wrong?

congoblue, 2019-01-03:
Sorry to update my own question, but it appears if I send "read" without the -f then it is working. Read by HTTP just seems to return the cached value, whether or not I send "maxage" or "required".  
But if I use a script to send the tcp read command periodically, I can then read the most recent value with HTTP which is good enough for my application.

john30, 2019-01-03:
please check the ebusd.log file for communication issues after issuing forced read

congoblue, 2019-01-04:
Hi John. When HTTP read is sent with "maxage" or "required", there is nothing shown in ebusd.log other than the normal commands going between boiler and control. Sending the command by TCP shows the "sent" command as in my top post.

john30, 2019-01-04:
maybe the message is only retrievable when updated by another slave, but not actually queriable

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30