john30/ebusd #249: Running ebusd in docker container without -f

peti69, 2019-01-05:
Hi,

I have deployed ebusd in docker (by means of portainer, no console docker access available). When using the option -f everything is fine but I do not get a log file on a volume (as documented). But if I start ebusd without -f docker/portainer gets confused somehow. Looks like that docker/portainer is not able to monitor ebusd properly if it is started as daemon. 

Kind regards,
Peter

john30, 2019-01-06:
you can't start it without "-f" in docker, thats why it is hard coded in the [Dockerfile](https://github.com/john30/ebusd/blob/859987ec89e1d819dec84a1eac2df1711f065b82/contrib/docker/Dockerfile#L35).
the log is usually retrieved from the docker daemon instead of the logfile (therefor "-f"). for dockerd, you can use e.g. "docker logs ebusd". So you'll need to find out, how portainer does that

closed 2019-01-06 by john30

peti69, 2019-01-06:
portainer does a "docker logs ebusd" and presents that on the GUI. That it, I have access to ebusd output,. But I would rather prefer to have it persistent on a volume for later investigation. Seems that there is no standard way (best practices) for managing application log files in conjunction with docker.

john30, 2019-01-06:
afaik you can make portainer use a different logging system than the default. check the documentation

peti69, 2019-01-07:
It is not possible for me to change the portainer setup. I only can use it.

john30, 2019-01-08:
I tried it and you just have to change the logging when adding the container

peti69, 2019-01-08:
What do you mean?

TheEPIX, 2019-01-08:
It's NOT possible to change a existing docker-image. ("Container" register)
It's only when you make the setup in the syno-GUI for the docker (in the "Image" register):
you klick on the "Start" Button and in the "environment"-register you filnd at the bottom a field where you can define the paramters for the docker-start..

Dont think too much - just try it ;-)