john30/ebusd #254: CMake: Test HAVE_DIRECT_FLOAT_FORMAT - Failed

andr2000, 2019-01-11:
Trying to cross-compile with cmake produces the following errors during configuration:
```
-- Performing Test HAVE_DIRECT_FLOAT_FORMAT
CMake Error: TRY_RUN() invoked in cross-compiling mode, please set the following cache variables appropriately:
   HAVE_DIRECT_FLOAT_FORMAT_EXITCODE (advanced)
For details see /home/a2k/projects/vaillant/sources/ebusd-build/TryRunResults.cmake
-- Performing Test HAVE_DIRECT_FLOAT_FORMAT - Failed
-- Configuring incomplete, errors occurred!
```
The issues seems to be that CMake tries to run target code on host and fails

john30, 2019-01-11:
the cmake is not desired to used for cross compilation yet. I wonder how one can determine availability of certain compiler features in such an environment then...

andr2000, 2019-01-12:
This is no problem when used with Yocto which is my case.
Yocto also provides cross-SDK, so there is no worry about the environment as well

john30, 2019-01-13:
but that does not solve the problem, right?

andr2000, 2019-01-14:
Exactly, because the CMakeLists is written in a way it requires the cross-compiled test to run on the host which is obviously not possible

john30, 2019-01-14:
exactly. so you can't cross compile right now, as said before

closed 2019-01-14 by john30

andr2000, 2019-01-14:
Hm, but it can be fixed.
I can cross-compile using autotools, but not CMake

john30, 2019-01-17:
if you can fix it, then please do so, as I wouldn't know how that should be possible