john30/ebusd #255: ebusd does not refresh data on MQTT /get or write data on MQTT /set

AloisKlingler, 2019-01-12:
Hi,

I am using ebusd 3.3 (ebusd-3.3_armhf-jessie_mqtt1.deb on a raspbian jessie with 1.3.4-2+deb8u3 MQTT version 3.1 client library) to fetch stuff from my brink device.
if the data is fetched if i use ebusctl read -f -n ... it works, but if I try to refresh it via MQTT /get or try to write data via MQTT /set I can only see the corresponding logentry, but ebusd does nothing.
I captured in wireshark the mqtt messages and they look valid (or even what I suppose to be valid).

I start ebusd with:
ebusd --loglevel=debug -d /dev/ttyUSB0 --configpath=/etc/ebusd/ -f --latency=20000 --mqtthost=localhost --mqttport=1883 --mqttlog --mqtttopic=ebusd/%circuit/%name/%field --accesslevel='*'

I need the latency flag otherwise it is not possible to read anything ebusd connects trough an USB to ebusd adapter (from mikrocontroller.net).

In /etc/ebusd/ there is only one file called "3c.csv"
a few lines of it:
## Renovent Excellent 300 series
*r,kwl,,,,3c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,kwl,,,,3c,4050,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,LuftmengeStufe1,,,,4050,01,aktuell,,UIR,,m³/h,,minimum,,UIR,,m³/h,,maximum,,UIR,,m³/h,,Schrittgroesse,,UIR,,m³/h,,Werkseinstellung,,UIR,,m³/h
r,,Aussenlufttemperatur,4022011A,,,4022,1a,,,SIR,10,°C,OptionTemperature

On looking with mosquitto_sub I can see:
mosquitto_sub -h localhost -t "ebusd/kwl/Aussenlufttemperatur/#"
7.0
get
get
get
7.6
7.4
7.3

or
heliotherm:~# mosquitto_sub -h localhost -t "ebusd/kwl/LuftmengeStufe1/#"
75
get

I tried:
mosquitto_pub -h localhost -t "ebusd/kwl/Aussenlufttemperatur/get" -m "get"
mosquitto_pub -h localhost -t "ebusd/kwl/LuftmengeStufe1/set" -m "75"
mosquitto_pub -h localhost -t "ebusd/kwl/LuftmengeStufe1/get" -m "get"
mosquitto_pub -h localhost -t "ebusd/kwl/LuftmengeStufe1/" -m "get"

logging looks like:
2019-01-12 17:11:23.057 [mqtt debug] log Client ebusd_3.3_8882 received PUBLISH (d0, q0, r0, m0, 'ebusd/kwl/LuftmengeStufe1/', ... (3 bytes))

or
2019-01-12 17:12:32.890 [mqtt debug] log Client ebusd_3.3_8882 received PUBLISH (d0, q0, r0, m0, 'ebusd/kwl/LuftmengeStufe1/get', ... (0 bytes))
2019-01-12 17:12:32.890 [mqtt debug] received topic ebusd/kwl/LuftmengeStufe1/get
2019-01-12 17:12:33.191 [main debug] performing regular tasks

is there anything what I forgot?

Thanks so far!

Best regards
Alois

andig, 2019-01-12:
That should work with `.../get`. Could you create a full ebusd log file including the MQTT operations?

andig, 2019-01-12:
Actually, you have `--mqtttopic=ebusd/%circuit/%name/%field` but your queries are different? What is %name?

AloisKlingler, 2019-01-12:
hi andig,

have copied the --mqtttopic from https://github.com/J0EK3R/ebusd-configuration-weishaupt/wiki/ebusd-mqtt-fhem - on trying to look how it is working. 

I adapted the queries regarding the tcpdump I made, but did not adapt the command line parameter.

Without this --mqtttopic=... it is working. :-)

Thanks!

Alois

closed 2019-01-13 by john30