john30/ebusd #261: Problem with read on VR 700

madito, 2019-02-10:


john30, 2019-02-10:
what do you mean with "I do not get any data from the VR700" exactly?
and what is "... it is updated from the JSON, but it is not updated" supposed to tell me?

madito, 2019-02-10:
Hi John,

The ebusd detect de VR700 but no read any parameters all marqued like unknow.

But I execute "ebusctl read" and read the value correct, and the json show this value. The json show nulls values if not update the values manually, executing the ebusctl on command line.

Is like if the ebusd not process the data of VR 700 "automaticaly"

madito, 2019-02-11:
You need the ebusd.log?? 

Thanksss

john30, 2019-02-14:
ebusd is not intended to query all messages by itself. however, you can change the config to make ebusd query within the pollinterval (see start parameters) and you can also dynamcially change the poll priority of messages during runtime.

closed 2019-02-14 by john30