john30/ebusd #263: Log rotate error

timstanley1985, 2019-02-11:
I am running ebusd on a raspberry pi 3b+ running raspbian.  Ebusd is great and works flawlessly but I get the annoying error message below once a day.  Removing copy truncate from the logrotate profile fixes this issue.  I will make a pull request now.

‘/etc/cron.daily/logrotate:
error: Ignoring ebusd because of bad file mode - must be 0644 or 0444.‘

john30, 2019-02-14:
what is the file mode of your /etc/logrotate.d/ebusd ?

timstanley1985, 2019-02-14:
It is 644

‘tim@pi3plus:/var/log$ ls -alh ebusd.log
-rw-r--r-- 1 root adm 27K Feb 14 11:15 ebusd.log’

john30, 2019-02-14:
not the log file, the logrotate file as stated above

john30, 2019-02-14:
the logrotate file mode was supposedly fixed with 859987ec89e1d819dec84a1eac2df1711f065b82

timstanley1985, 2019-02-14:
Ah, I thought it was complaining about the log file, not the logrotate profile file.  I am running a packaged version of ebusd so it doesn't include this fix.  I will change permissions manually and wait for next version.

Sorry to trouble you with this

closed 2019-02-14 by timstanley1985