john30/ebusd #265: Home Assistant ebusd component

Dinth, 2019-02-27:
That's not really an issue, but maybe it would be worth mentioning that on ebusd website, as since the last HA release, it does support ebusd natively. For now it's only reading predefinied messages, but hopefully the developer will keep working on it.

john30, 2019-02-28:
can you point out a site where this is described in more detail? then I would add that to the [integrations page in the wiki](https://github.com/john30/ebusd/wiki/7.-Integrations)

Dinth, 2019-02-28:
Sure John, here it is: https://www.home-assistant.io/components/ebusd/

john30, 2019-03-09:
ok, I've added it to the wiki. could you provide a screen shot or something to get an impression of it?

andig, 2019-03-21:
@john30 could I ask you how you've developed the technical drawing (and sensor values) on the integrations page? Is that part of fhem? Any special component?

Dinth, 2019-03-21:
> ok, I've added it to the wiki. could you provide a screen shot or something to get an impression of it?

I believe that @CrazYoshi can provide better screenshots than i do, as he's the developer and his system is BAI is more supported by this integration than mine.

john30, 2019-03-23:
> 
> 
> @john30 could I ask you how you've developed the technical drawing (and sensor values) on the integrations page? Is that part of fhem? Any special component?

this was my attempt to create a common way of providing a UI showing the flows, temperatures, pressures and so on. it is part of ebusd actually if you enable the http port and make it use the html files from https://github.com/john30/ebusd/tree/master/contrib/html, but it is currently limited to ehp based setups only. for others it would require a lot of config files like those in the https://github.com/john30/ebusd/tree/master/contrib/html/items folder and my plan was to move all that stuff directly into v3 config files on ebusd-configuration. In parallel I started a web frontend for editing config definitions, but I'm lacking time for both sub-projects unfortunately.

andig, 2019-11-04:
> That's an issue, but maybe it would be worth mentioning that on ebusd website

> ok, I've added it to the wiki. 

Looks like this can be closed?

closed 2019-11-05 by john30