john30/ebusd #267: Docker Image not run on Raspberry Pi 3

SandnerSoft, 2019-02-28:


john30, 2019-03-01:
no idea, try google

dkuschmierz, 2019-11-04:
Hi @SandnerSoft,

your script is almost working, but you forgot to set the right FROM ...

Here is a for me working Dockerfile:

`FROM balenalib/raspberrypi3

RUN apt-get update && apt-get install -y \
    logrotate libmosquitto1 libstdc++6 libc6 libgcc1 \
    curl \
    && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt-get install -y \
    autoconf automake g++ make git \
    libmosquitto-dev \
    && rm -rf /var/lib/apt/lists/*

LABEL maintainer "ebusd@ebusd.eu"

ENV EBUSD_VERSION 3.4
ENV EBUSD_ARCH armhf
ENV UDEV=1

LABEL version "${EBUSD_VERSION}-${EBUSD_ARCH}-devel"

WORKDIR /build

RUN (curl -SL https://github.com/john30/ebusd/archive/master.tar.gz \
    | tar xz --strip-components=1) \
    && ./autogen.sh \
    && make install-strip

WORKDIR /

RUN rm -rf /build

EXPOSE 8888

COPY release/docker-entrypoint.sh /
ENTRYPOINT ["/docker-entrypoint.sh"]
CMD ["-f", "--scanconfig"]`

@john30 
Would it be possible to build also this Docker-Image when you release new versions. Currently I change my home-assistant installation completely to docker-images and this one is missing

commit 3e7e4c7a (2019-11-10): add RPi version (fix #267)
closed 2019-11-10 by john30