john30/ebusd #268: Not working at all. Any hint?

EUA, 2019-03-05:
Using OrangePi Zero + PL2303 USB cable with armbian OS.
Fetched from git and compiled successfully. But can't make it work.

Boiler is : Vaillant turboMAX plus 800 

`ebusd  --scanconfig=full -f -D  -d /dev/ttyUSB0 --latency=1000`
returns
```
root@orangepizero:~/ebusd/src# ebusd  --scanconfig=full -f -D  -d /dev/ttyUSB0 --latency=1000
2019-03-05 06:17:56.821 [main notice] ebusd 3.3.v3.3-13-gd366bbb started with full scan
2019-03-05 06:17:57.672 [bus notice] bus started with own address 31/36
2019-03-05 06:20:07.065 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
```
ebusctl i
```
version: ebusd 3.3.v3.3-13-gd366bbb
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

```

I can verify that the USB to serial cable is working via sending raw signals to serial port using python:
```
import serial
s = serial.Serial('/dev/ttyUSB0')
w=b'\x07\x02\x00\x00\x00\x04\xc4'
s.write(w)
s.read_all()
```
returns me
'\x08\x00\x00\x93\x0b\t\xfb\x0b'

I can also fetch data via vrDialog program using windows and the USB-to-TTL cable.

What can I do more?

john30, 2019-03-05:
does `ebusd --scanconfig=full -f -D -d /dev/ttyUSB0 --latency=1000 --lograwdata=bytes` show any `>`/`<` entries?

EUA, 2019-03-05:
Nope. No such output observed. My device has "VUW TR 242/2-5 R1" label might be not compatible or old...

I made WiFi-to-Serial device and captured TCP trafic via wireshark to get communication dump of vrDIALOG/2 software. Than filtered the TCP stream.
I believe you can easily diagnose if this stream is compatible with your stack.
Answers from ebus start with spaces, Since queries and answers are in multiple TCP packed, dump look little weird but not hard to understand the stream I think.)

```
00000000  07                                               .
00000001  02 00 00 00 04 c4                                ......
    00000000  08                                               .
    00000001  00                                               .
    00000002  00                                               .
    00000003  93 0b                                            ..
    00000005  09                                               .
    00000006  fb 0b                                            ..
00000007  07                                               .
00000008  00 00 00 16 00 cc                                ......
    00000008  06                                               .
    00000009  00                                               .
    0000000A  ff                                               .
    0000000B  21 aa                                            !.
    0000000D  5f                                               _
0000000E  07                                               .
0000000F  00 00 00 14 00 c8                                ......
    0000000E  08                                               .
    0000000F  00                                               .
    00000010  0c                                               .
    00000011  54 f3                                            T.
    00000013  ab                                               .
    00000014  aa 62                   
```
Full log is here:
https://pastebin.com/PXuqABRh

EUA, 2019-03-09:
Sorry. I think this is not EBUS but X6 protocol.
Thanks.

closed 2019-03-09 by EUA

john30, 2019-03-09:
the thermocompact seems to have a RJ-45 service connector though on the electronic board as of page 32 of the [manual](https://www.freeboilermanuals.com/assets/pdf/vaillant/thermoCOMPACT.pdf). this is usually eBUS, but not sure about it.

EUA, 2019-03-11:
Yes, I use that 6P6C RJ-11 port for communicate. My signal is RS232-TTL +5v compatible.
Might be commands are compatible with EBUS too but your stack doesn't work on it somehow.

One thing that I need to tell, sometime boiler doesn't respond to first commands.
Look like in a kind of sleeping mode. After repeating same signal (2.nd, 3.rd or at 4.th...), than boiler responds and keep responding each commands without issue.

john30, 2019-03-16:
no more ideas here