john30/ebusd #270: bus error signal lost

CrazYoshi, 2019-03-10:
Hi,
I have issues running the last version of ebusd on my orangepi.
If I look at the log file i see a lot of entries like this one:
`2019-03-10 16:06:53.032 [bus error] signal lost`
but if I execute ebusctl info command I see that connection is established:
`pi@homedaemon:~$ ebusctl info
version: ebusd 3.3.v3.3-13-gd366bbb
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/15.700.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 12
max symbol rate: 171
min arbitration micros: 40
max arbitration micros: 107
min symbol latency: 6
max symbol latency: 17
reconnects: 0
masters: 3
messages: 611
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=1303", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd`

Morover I cannot even navigate to my ebus adapter administrator panel when ebusd service is active but at soon as I stop it the page is available again. 
How is it possibile that version 3.3 force the adapter to freeze?

andig, 2019-09-16:
Seems this can be closed after 6 months?

john30, 2019-09-17:
agreed

closed 2019-09-17 by john30