john30/ebusd #271: MQTT doesn't receive any update from VRC700

CrazYoshi, 2019-03-12:
Hi,
I have the following issue using ebusd and MQTT.
I'm currently using this configuration:
`EBUSD_OPTS="-d 192.168.1.23:8889 -l /var/log/ebusd.log --loglevel=error --latency=50000 --scanconfig --mqtthost=localhost --mqttport=1883 --mqttuser=homeassistant --mqttpass=Gr1sul0n386 --mqttjson --mqttretain --mqttversion=3.1.1 --accesslevel="*""`
From MQTT.fx log I see lot of messages coming from bai circuit but any coming from 700 unless I try to read or write using MQTT or ebusctl commands.
My ebusctl info is the following:
`pi@homedaemon:~$ ebusctl info
version: ebusd 3.3.v3.3-13-gd366bbb
access: *
signal: acquired
symbol rate: 89
max symbol rate: 93
min arbitration micros: 16
max arbitration micros: 211
min symbol latency: 10
max symbol latency: 69
reconnects: 0
masters: 3
messages: 611
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=1303", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
Do you have any clue on this? I would expect to receive lot of change about inside temperature and so on..

john30, 2019-03-16:
thats normal, as no circuit actually queries the 700 for some data. it is indeed the 700 querying the bai and that's why you get data for the bai frequently. the 700 might send some info about temperature by broadcast, but i don't know.

CrazYoshi, 2019-03-16:
But if I query 700 from another source via TCP and receive data there, why should I need MQTT? 
It seems pretty useless to me MQTT used in this way..
Which should be the use case in order to use MQTT?

Dinth, 2019-03-16:
> But if I query 700 from another source via TCP and receive data there, why should I need MQTT?
> It seems pretty useless to me MQTT used in this way..
> Which should be the use case in order to use MQTT?

Hi CrazYoshi. It seemed weird for me as well, but that's the way it ebusd works.
When you read a variable through ebusctl or TCP, ebusd actually queries the device what's the value at the moment. When you read MQTT variable though, ebus doesnt know if when are you trying to read the var, so it cannot query the device for it. 
At least this is how i understood this.
Anyhow, you can either either force read of particular value by submitting a message to variable/got topic, or alternatively you can set up cron to automatically run supplied script to refresh all the values every 7 minutes or so.

john30, 2019-03-17:
that's not correct. sure you can issue a read of a message through MQTT on ebusd as [explained here](https://github.com/john30/ebusd/wiki/3.3.-MQTT-client).

closed 2019-03-17 by john30

CrazYoshi, 2019-03-19:
> that's not correct. sure you can issue a read of a message through MQTT on ebusd as [explained here](https://github.com/john30/ebusd/wiki/3.3.-MQTT-client).

It can be not correct in your opinion but you don't explain anything and you don't have answer to my questions. Which should be the use case in order to use MQTT?

andig, 2019-03-19:
> Which should be the use case in order to use MQTT?

@CrazYoshi not sure what you mean. I'm using it all the time to read/write from node-red.

CrazYoshi, 2019-03-19:
> > Which should be the use case in order to use MQTT?
> 
> @CrazYoshi not sure what you mean. I'm using it all the time to read/write from node-red.

How you trigger ebus reads? What I don't understand is why should I trigger reads using TCP/MQTT that also most of the time if I do one immediately after the other one goes in error. Why ebusd can automatically triggers reads?

andig, 2019-03-19:
> How you trigger ebus reads?

Exactly as documented: `ebusd/700/fancyregister/get`

> What I don't understand is why should I trigger reads using TCP/MQTT that also most of the time if I do one immediately after the other one goes in error. Why ebusd can automatically triggers reads?

Again- I don't understand what you mean. ebusd can do polling but it does not poll all registers. So by default you will only see traffic that already *is* on the bus. If you want more you need to query the bus.

john30, 2019-03-23:
@CrazYoshi MQTT was added in order to ease integration into other projects since it is standardized. if you don't want to use it, then don't. I really don't get your point here.