john30/ebusd #272: ebusd and tado - any experiences  ?

stefaanv, 2019-03-13:
I've added Tado radiator knobs and thermostats to my heating setup.  They seem to have better control over the Vaillant burner than my old Vaillant thermostat itself.  Lot's of unknow messages in the ebusd logs.

Anyone have experience with tado+ebusd ?

CrazYoshi, 2019-03-14:
Hi,
as you can see here [https://github.com/john30/ebusd/wiki/6.-Hardware](url) ebusd has no support for Tado.
That can be controlled directly or using software like homeassistant.

Dinth, 2019-03-14:
> I've added Tado radiator knobs and thermostats to my heating setup. They seem to have better control over the Vaillant burner than my old Vaillant thermostat itself. Lot's of unknow messages in the ebusd logs.
> 
> Anyone have experience with tado+ebusd ?

Can Tado even speak ebus? Is ebusd connected to your boiler and can you see ebus messages with Tado talking to your boiler?

stefaanv, 2019-03-14:
The extension kit is connected to the ebus of the burner and is talking to it, I see the messages but they're unknown.  I'll try to decode some, it will take some time, I'm new to this.

andig, 2019-11-04:
It seems the tado capabilties are very limited: https://community.tado.com/en-gb/discussion/657/support-current-and-more-recent-vaillant-ebus-controllers. If tado were to partner with ebusd the reverse engineering efforts could be combined...

stefaanv, 2019-11-16:
Same experience here.
I have tado on my Vaillant burner and monitor it with ebusd.
All tado is doing is switching the old thermostat on and off and setting the flow temperature.  There're actually doing that in a really crappy way 
- setting the flow temperature way to high - up to 70° when outside temperature is 10°
- asking heat from the burner while closing all the radiators
- swing in room temperature is high, I see overshoots of up to 1° over set temperature
With this control strategy, their claim to save energy is absolute nonsense