john30/ebusd #273: Problem with getting telegrams

create1st, 2019-03-22:
My setup is RPi3B+, eSera eBus to serial Coupler, USB ADS-1PQ (FTDI FT232RL + ZT213LEEA), Wolf CGS-20 (from my understanding it is based on Kromschröder ebus)
I'm not quite sure what kind of issue I'm having here. Once I start in scanconfig / scanconfig=full I don't get anything
`
$ sudo ebusd -l --log="all trace" -f --device=/dev/ttyUSB0 --scanconfig
2019-03-22 22:46:32.237 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-03-22 22:46:32.788 [bus notice] bus started with own address 31/36
2019-03-22 22:46:32.800 [bus notice] signal acquired
2019-03-22 22:48:42.907 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
`
Also ebusctl infor does not give me a good insight
`
$ ebusctl info
version: ebusd 3.3.v3.3
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
signal: acquired
symbol rate: 32
max symbol rate: 61
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
Output from raw data give such a thing which to me looks okay but maybe my experience is to low to identify the problem here. I was checking the output and I'm getting the whole range of values (does it means that potentiometer adjustment is fine?).

`
$ ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0
2019-03-22 22:58:19.478 [bus notice] bus started with own address 31/36
2019-03-22 22:58:19.509 [bus notice] <d5
2019-03-22 22:58:19.509 [bus notice] signal acquired
2019-03-22 22:58:19.554 [bus notice] <ff
2019-03-22 22:58:19.597 [bus notice] <aa
2019-03-22 22:58:19.687 [bus notice] <ba
2019-03-22 22:58:19.777 [bus notice] <eb
2019-03-22 22:58:19.823 [bus notice] <c9
2019-03-22 22:58:19.867 [bus notice] <aa
2019-03-22 22:58:19.913 [bus notice] <ab
2019-03-22 22:58:19.958 [bus notice] <f5
2019-03-22 22:58:20.003 [bus notice] <dd
2019-03-22 22:58:20.094 [bus notice] <fd
2019-03-22 22:58:20.138 [bus notice] <bb
2019-03-22 22:58:20.182 [bus notice] <ab
2019-03-22 22:58:20.228 [bus notice] <ba
2019-03-22 22:58:20.272 [bus notice] <ae
2019-03-22 22:58:20.320 [bus notice] <f5
2019-03-22 22:58:20.362 [bus notice] <aa
2019-03-22 22:58:20.408 [bus notice] <dd
2019-03-22 22:58:20.543 [bus notice] <ab
2019-03-22 22:58:20.588 [bus notice] <ba
2019-03-22 22:58:20.633 [bus notice] <d5
2019-03-22 22:58:20.678 [bus notice] <dd
2019-03-22 22:58:20.725 [bus notice] <f5
2019-03-22 22:58:20.769 [bus notice] <f5
2019-03-22 22:58:20.860 [bus notice] <ff
2019-03-22 22:58:20.904 [bus notice] <d5
2019-03-22 22:58:20.949 [bus notice] <ba
2019-03-22 22:58:20.993 [bus notice] <bf
2019-03-22 22:58:21.037 [bus notice] <ab
2019-03-22 22:58:21.129 [bus notice] <d5
2019-03-22 22:58:21.173 [bus notice] <af
2019-03-22 22:58:21.219 [bus notice] <dd
2019-03-22 22:58:21.263 [bus notice] <ea
2019-03-22 22:58:21.308 [bus notice] <ee
2019-03-22 22:58:21.355 [bus notice] <fd
2019-03-22 22:58:21.400 [bus notice] <fd
2019-03-22 22:58:21.443 [bus notice] <ab
2019-03-22 22:58:21.488 [bus notice] <d5
[... 124 lines cut ...]
2019-03-22 22:58:25.941 [bus notice] <60
2019-03-22 22:58:25.945 [bus notice] <51
2019-03-22 22:58:25.949 [bus notice] <2a
2019-03-22 22:58:25.953 [bus notice] <0f
2019-03-22 22:58:25.958 [bus notice] <07
2019-03-22 22:58:25.963 [bus notice] <c2
2019-03-22 22:58:25.966 [bus notice] <32
2019-03-22 22:58:25.971 [bus notice] <5f
2019-03-22 22:58:25.975 [bus notice] <7e
2019-03-22 22:58:26.024 [bus notice] <ab
2019-03-22 22:58:26.068 [bus notice] <aa
2019-03-22 22:58:26.115 [bus notice] <ff
2019-03-22 22:58:26.161 [bus notice] <fd
2019-03-22 22:58:26.248 [bus notice] <aa
`

I have to say that I was 2 times lucky with getting something which was even identified as Kromschröder  but I'm not convinced that it was not by accident. First time I have got 4 masters and 1 slave (plus additional ebusd master and slave). This however looks to me was all the wya wrong as I do have 1 central unit and one wall controller. The other time it was more than good but I was getting
2019-03-17 09:54:19.255 [update error] unable to parse update-read broadcast datetime from 30fe070009008011061000000700 / : ERR: argument value out of valid range

and then I have lost this completely once I have tried to do some adjustments with potentiometer

What am I doing wrong or does the output from raw data gives some insight where is the problem?

Thanks,
Sebastian

john30, 2019-03-23:
you basically answered yourself: the poti setting is not good enough anymore. you need to adjust it exactly to habe the right bytes arriving. if you add a higher log level youo should also see plenty of CRC mistakes and invalid lengths and stuff. all of this has to vanish besides rather rare exceptions.

create1st, 2019-03-23:
Hi @john30,
Thanks. That was more than helpful. I was not aware that I can have more insightful output. I'll try again later Today or Tomorrow.

create1st, 2019-03-26:
Unfortunately I was not lucky. Adjusting the potentiometer  did not get me any result. Once the potentiometer is at the max. I'm getting
```
2019-03-23 11:17:52.916 [bus debug] ERR: invalid escape sequence during receive command, switching to skip
2019-03-23 11:17:54.003 [bus notice] max. symbols per second: 234
2019-03-23 11:17:55.134 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:17:55.138 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:01.393 [main debug] performing regular tasks
2019-03-23 11:18:07.530 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:07.534 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:11.394 [main debug] performing regular tasks
2019-03-23 11:18:17.546 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:17.550 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:21.394 [main debug] performing regular tasks
2019-03-23 11:18:26.785 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:26.788 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:31.394 [main debug] performing regular tasks
2019-03-23 11:18:34.660 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:34.665 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:38.501 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:38.504 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:38.958 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2019-03-23 11:18:38.962 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2019-03-23 11:18:41.395 [main debug] performing regular tasks
```
Once I'm moving it to min. level then I'm getting timeouts.
```
2019-03-23 11:21:11.701 [main debug] performing regular tasks
2019-03-23 11:21:11.745 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:12.014 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:12.328 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:12.588 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:12.855 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:13.128 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:13.267 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:13.580 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:13.778 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:13.957 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:14.813 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:14.903 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:15.021 [bus debug] ERR: read timeout during receive command, switching to skip
2019-03-23 11:21:15.829 [bus debug] ERR: read timeout during receive command, switching to skip
```
No stable place in whole range. So I ended up in the conclusion that one of the elements of my setup is not working correctly. My biggest suspect is that RPI is not able to power properly RS-232 to USB cable. 
Once I will have some more spare time I will try to install ebusd on Windows and try with desktop. Any other suggestion which can help me diagnose the problem?

john30, 2019-03-27:
see [my comment here](https://github.com/john30/ebusd/issues/274#issuecomment-476989045)

john30, 2021-10-06:
closed due to inactivity