john30/ebusd #275: Set dev path when autostarting with systemctl

djmvt, 2019-03-30:
Been testing with ebusd and home build adapter. The adapter works when I connect it as a single device on my Raspberry pi but when I also connect my p1 serial to USB and homeduino to my Raspberry pi the ttyUSB mount gets switched. I can do a manual start with - d option to set the dev/ttyUSB path but how to do this when autostarting ebusd? And preferably I would like to set the autostart mount point by name rather than ttyUSBx because I've had some issues with that in the past (mostly because of cheap China serial to usb converters with the same name/vendor)

john30, 2019-03-31:
just like many other daemons this has to be configured in /etc/default/ebusd on Debian systems

closed 2019-03-31 by john30

djmvt, 2019-03-31:
Thanks! Learning more and more ;)