john30/ebusd #276: raspberry pi

goosst, 2019-03-30:
Hello

Thank you for all the work around ebusd. I have some issue with the raspberry pi:

I'm using V2.2 of the adapter hardware (connected through USB using a cp210).
Is connected to a vaillant heater.

On an old pc running MKlinux the output seems -as far as I can tell - correct:

> 2019-03-30 10:42:59.873 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:05.895 [update notice] received read bai Status01 QQ=10: 56.5;56.5;-;45.0;50.0;off
> 2019-03-30 10:43:07.865 [update notice] received update-read broadcast vdatetime QQ=10: 10:43:54;30.03.2019
> 2019-03-30 10:43:08.106 [update notice] received unknown MS cmd: 1008b512020064 / 00
> 2019-03-30 10:43:09.893 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:15.913 [update notice] received read bai Status01 QQ=10: 56.5;56.5;-;45.0;50.0;off
> 2019-03-30 10:43:17.098 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;-
> 2019-03-30 10:43:17.915 [update notice] received read bai Status02 QQ=10: on;60;75.0;70;65.0
> 2019-03-30 10:43:19.962 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:25.936 [update notice] received read bai Status01 QQ=10: 56.0;56.0;-;45.0;50.0;off
> 2019-03-30 10:43:27.926 [update notice] received update-read broadcast outsidetemp QQ=10: 0.000
> 2019-03-30 10:43:29.970 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:35.989 [update notice] received read bai Status01 QQ=10: 56.0;56.0;-;45.0;50.0;off
> 2019-03-30 10:43:37.976 [update notice] received read bai Status02 QQ=10: on;60;75.0;70;65.0
> 2019-03-30 10:43:40.022 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:46.037 [update notice] received read bai Status01 QQ=10: 56.0;55.5;-;45.0;50.0;off
> 2019-03-30 10:43:50.029 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> 2019-03-30 10:43:56.048 [update notice] received read bai Status01 QQ=10: 55.5;55.5;-;45.0;50.0;off
> 2019-03-30 10:44:00.081 [update notice] received update-write bai SetMode QQ=10: auto;52.5;-;-;0;0;1;0;0;0
> ^C2019-03-30 10:44:02.068 [main notice] SIGINT received
> 2019-03-30 10:44:05.713 [main notice] ebusd stopped
> 

I conclude from this the adapter works as expected.
However, when I plug the adapter into USB port of a raspberry pi 3, I get the following behaviour. 
> pi@raspberrypi:~ $ ebusd -f --scanconfig
> 2019-03-30 10:57:49.434 [main notice] ebusd 3.3.v3.3 started with auto scan
> 2019-03-30 10:57:49.907 [bus notice] bus started with own address 31/36
> 2019-03-30 10:57:49.910 [bus notice] signal acquired
> 2019-03-30 10:57:50.273 [bus notice] new master 10, master count 2
> 2019-03-30 10:57:50.273 [update notice] received unknown BC cmd: 10feb516080018011130030619
> 2019-03-30 10:57:50.511 [bus notice] new master 03, master count 3
> 2019-03-30 10:57:50.511 [update notice] received unknown MS cmd: 1008b512020064 / 00
> 2019-03-30 10:57:52.299 [update notice] received unknown MS cmd: 1008b51009000035ffffff44ff00 / 0101
> 2019-03-30 10:57:58.319 [update notice] received unknown MS cmd: 1008b5110101 / 0960610080555f0000ff
> 2019-03-30 10:58:00.045 [bus notice] scan 08: ;Vaillant;BAI00;0202;9602
> 2019-03-30 10:58:00.045 [update notice] store 08 ident: done
> 2019-03-30 10:58:00.045 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0202;9602
> 2019-03-30 10:58:00.045 [bus notice] scan 08: ;Vaillant;BAI00;0202;9602
> 2019-03-30 10:58:00.309 [update notice] received unknown MS cmd: 1008b5110102 / 06013c9646826e
> 2019-03-30 10:58:00.375 [main notice] read common config file vaillant/scan.csv
> 2019-03-30 10:58:00.453 [main notice] read common config file vaillant/general.csv
> 2019-03-30 10:58:00.535 [main notice] read common config file vaillant/broadcast.csv
> 2019-03-30 10:58:00.620 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0202, HW9602
> 2019-03-30 10:58:00.786 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 10:58:00.966 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 10:58:01.147 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 10:58:01.327 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
> 2019-03-30 11:00:42.481 [main notice] found messages: 198 (2 conditional on 24 conditions, 0 poll, 9 update)
> 2019-03-30 11:00:42.606 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 11:00:42.788 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 11:00:42.972 [update notice] sent scan-read scan.08 id QQ=31: 
> 2019-03-30 11:00:43.152 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
> 2019-03-30 11:00:43.153 [bus notice] scan 08: ;21;17;09;0010011632;1300;378112;N5
> 2019-03-30 11:00:57.051 [main error] scan config 15: ERR: no signal
> 2019-03-30 11:00:57.052 [bus error] signal lost
> 2019-03-30 11:00:57.091 [bus notice] signal acquired
> 2019-03-30 11:01:28.038 [main error] scan config 15: ERR: no signal
> 2019-03-30 11:01:28.039 [bus error] signal lost
> 2019-03-30 11:01:28.158 [bus notice] signal acquired
> ^C2019-03-30 11:01:49.161 [main notice] SIGINT received
> 2019-03-30 11:01:53.215 [main notice] ebusd stopped
> pi@raspberrypi:~ $ ebusd -f --scanconfig
> 2019-03-30 11:01:55.584 [main notice] ebusd 3.3.v3.3 started with auto scan
> 2019-03-30 11:01:56.041 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
> 2019-03-30 11:01:56.042 [bus notice] bus started with own address 31/36
> 2019-03-30 11:01:56.042 [bus notice] device invalid
> 2019-03-30 11:02:01.047 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
> 2019-03-30 11:02:01.047 [bus notice] device invalid
> 2019-03-30 11:02:06.051 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
> 2019-03-30 11:02:06.051 [bus notice] device invalid
> ^C2019-03-30 11:02:08.163 [main notice] SIGINT received
> ^C2019-03-30 11:02:08.550 [main notice] SIGINT received
> 

I've tried different power supplies for the raspberry without success. 
Is this an already known issue?

p.s: are there somewhere just some easy examples to find on read/write commands :)? It can be a bit overwhelming at first.

thanks!

andig, 2019-03-30:
Are you sure that the adapter is actually assigned to `/dev/ttyUSB0` on the raspi?

goosst, 2019-03-30:
I believe so:

this is when plugging the adapter in the raspberry pi:
```
pi@raspberrypi:~ $ dmesg | grep cp210
[    3.634437] usbcore: registered new interface driver cp210x
[    3.634513] usbserial: USB Serial support registered for cp210x
[    3.634655] cp210x 1-1.3:1.0: cp210x converter detected
[    3.645437] usb 1-1.3: cp210x converter now attached to ttyUSB0
pi@raspberrypi:~ $ ebusd -f --scanconfig
2019-03-30 16:54:22.996 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-03-30 16:54:23.314 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:23.314 [bus notice] bus started with own address 31/36
2019-03-30 16:54:23.314 [bus notice] device invalid
2019-03-30 16:54:28.319 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:28.319 [bus notice] device invalid
2019-03-30 16:54:33.324 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:33.324 [bus notice] device invalid
2019-03-30 16:54:38.329 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:38.329 [bus notice] device invalid
2019-03-30 16:54:43.334 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:43.334 [bus notice] device invalid
^C2019-03-30 16:54:45.113 [main notice] SIGINT received
2019-03-30 16:54:48.339 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-03-30 16:54:48.340 [bus notice] device invalid
2019-03-30 16:54:48.416 [main notice] ebusd stopped
```

directly after that, I plugged it back into my pc:
```
$ ebusd -f --scanconfig
2019-03-30 16:55:21.866 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-03-30 16:55:22.312 [bus notice] bus started with own address 31/36
2019-03-30 16:55:22.329 [bus notice] signal acquired
2019-03-30 16:55:26.636 [bus notice] new master 10, master count 2
2019-03-30 16:55:26.696 [bus notice] new master 03, master count 3
2019-03-30 16:55:26.696 [update notice] received unknown MS cmd: 1008b5110101 / 092e2f0080292c0000ff
2019-03-30 16:55:28.682 [update notice] received unknown MS cmd: 1008b5110102 / 06013c9646826e
2019-03-30 16:55:30.728 [update notice] received unknown MS cmd: 1008b51009000000ffffff45ff00 / 0101
2019-03-30 16:55:32.451 [bus notice] scan 08: ;Vaillant;BAI00;0202;9602
2019-03-30 16:55:32.451 [update notice] store 08 ident: done
2019-03-30 16:55:32.451 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0202;9602
2019-03-30 16:55:32.451 [bus notice] scan 08: ;Vaillant;BAI00;0202;9602
2019-03-30 16:55:32.793 [main notice] read common config file vaillant/scan.csv
2019-03-30 16:55:32.871 [main notice] read common config file vaillant/general.csv
2019-03-30 16:55:32.943 [main notice] read common config file vaillant/broadcast.csv
2019-03-30 16:55:33.019 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0202, HW9602
2019-03-30 16:55:33.136 [update notice] sent scan-read scan.08 id QQ=31: 
2019-03-30 16:55:33.304 [update notice] sent scan-read scan.08 id QQ=31: 
2019-03-30 16:55:33.471 [update notice] sent scan-read scan.08 id QQ=31: 
2019-03-30 16:55:33.640 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:33.919 [main notice] found messages: 198 (2 conditional on 24 conditions, 0 poll, 9 update)
2019-03-30 16:55:34.006 [bus notice] max. symbols per second: 102
2019-03-30 16:55:34.064 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:34.232 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:34.401 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:34.569 [update notice] sent scan-read scan.08 id QQ=31: 21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:34.569 [bus notice] scan 08: ;21;17;09;0010011632;1300;378112;N5
2019-03-30 16:55:36.719 [bus notice] scan 15: ;Vaillant;F3700;0114;6102

```

goosst, 2019-03-31:
I can also confirm the same behaviour (i.e. not functioning) on my ubuntu pc. So far it works on one laptop/OS, not on raspberry pi and not on my ubuntu-laptop.

john30, 2019-03-31:
this is not related to the adapter itself but to the serial to usb converter.
are you sure that no other tool is using the same device? e.g. fhem might use /dev/ttyUSB0 for some purpose or any other tool might as well. you can use "lsof" as root to check if any other tool uses the device as well

goosst, 2019-03-31:
I don't understand it anymore, did:
- a fresh install of raspbian lite on raspberry pi 3
- only installed ebusd

first boots it gave the same behaviour as before, now it suddenly started working consistently (over a reasonable time period and with two different power supplies ... )

thanks for the help :), I guess this one can be closed.

john30, 2019-04-01:
as said, you probably had another tool using the device which blocks ebusd from using it

closed 2019-04-01 by john30