john30/ebusd #279: Protherm Tiger Condens how to set heating

vajkooo, 2019-04-04:
Dear I would like to ask you for help with mqtt command of "start heating".
boiler: Protherm Tiger Condens
thermostat: Thermolink RC 

I succsessfully read information via MQTT from the boiler but I want to make next step and it's writing command for heating. 

How can I do it via MQTT? What parameter should be used for that? 

Some information from ebusd :

pi@pi-kotol:~ $ ebusctl info
version: ebusd 3.3.v3.3-18-g82ae848
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 42
max symbol rate: 123
min arbitration micros: 716
max arbitration micros: 858
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 3
messages: 212
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0607;HW=3302", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7f_2;SW=0001;HW=8002"
address 31: master #8, ebusd
address 36: slave #8, ebusd
  

Could somebody help me with that, please?

Thank you.

john30, 2019-04-12:
I don't think there is a direct way of doing this. instead you'd have to tell the controller (E7F) to do that, e.g. by increasing the desired temperature or adjusting the heat curve

vajkooo, 2019-04-12:
john30 thank you for your reply. But is it possible tell the controller E7F via MQTT command or I have to do it through coplicated way? 
I don't know what I'm doing wrong but I can't read any information from e7f controller. 
For example if i use command ebusctl read -f -v Time the answer is ERR: element not found
Is there any special way how to do that? 

Thank you again for your help

vajkooo, 2019-04-18:
I'm able to read from e7f_2 controller but which parameter should i use to for heating? I tried chande RoomTempDesired but wirte was finished with done, I saw it there with ebusctl find -w but it didn't changed on controller. 
Could anybody help me please?
Thanks

john30, 2019-04-19:
you should rather try with HeatingCurve

vajkooo, 2019-04-19:
Thank you for your reply. I'm working on it few days and I'm still not able send heating request. 
If I try read HeatingCurve then:
pi@pi-kotol:~ $ ebusctl r -f HeatingCurve
ERR: invalid position in decode

If i look correct the HeatingCurve is datatype curve but in _tepmplates.csv file this datatype is not defined. What type should be the HeatingCurve?
Thank you

john30, 2019-05-01:
in this case it seems you controller does not support this message.
you can find the actual data type being used by issuing a `ebusctl find -f heatingcurve`

vajkooo, 2019-05-01:
thak you for reply, but how can I send heating request if the heating curve is not supported?

john30, 2019-06-02:
you could try with some of the standard messages like b51000, but I'd rather not do it this way.
can you initiate a manual start of heating using the controllers UI? if yes, then you'd have to find the register in the controller that is used for that. if not, it's complicated...

samm-git, 2019-08-06:
I have a very similar configuration. I started to do CSV for the 0x15 device, which is RC controller. So far i been able to identify such values (vaillant/15.e7f_2.csv):

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,E7f_2,ExaCONTROL E7 radio,E7f00 206 001+101,,,,,,,,,,
*r,e7f,,,,"15","B509","0D",,,,,,
*w,e7f,,,,"15","B509","0E",,,,,,
*wi,e7f#install,,,,"15","B509","0E",,,,,,

r,,RoomTemp,,,,,"3A00",,,temp,,,Room Tempearature
r;w,,TargetRoomTemp,,,,,"2400",,,temp,,,Target Room Tempearature
r;w,,HWTemp,,,,,"4000",,,temp,,,HW Tempearature
r,,DevName,,,,,"3000",,,NTS:14,,,Thermostat name
```
good catch is that you can change hot water temperature this way. However, TargetRoomTemp so far is not changing, not sure why, but it is read only.

My assumption is that without some service docs i will not get much further. Also on some commands 0x15 may hang to the state where full restart is needed.

@john30 problem is that device 0x15 (0x10 master) is not really showing any internal flow, e.g. i mean that boiler is never telling what to do, all initiated from controller. And commands of controller are unknown.

john30, 2019-08-11:
hm, too bad @samm-git. thanks for sharing your findings anyway!

samm-git, 2019-08-11:
@john30 i been able to find a commends which changing parameters on RC controller :) 

Only problem here is that _sometime_ RF unit changing them back,  but sometime it is getting it from controller. I think they are speaking with each other in some kind of batches (i found yesterday with rtl-sdr that it is 866Mhz transfer). So probably my "chip and dirty" solution will be to setup script which lets say enforce my mode in a 10 minutes or so by sending it to controller.

samm-git, 2019-08-13:
@vajkooo i did some progress on this, see https://github.com/john30/ebusd-configuration/pull/140. Please let me know if it works for you.

vajkooo, 2019-11-04:
@samm-git - sorry for my late response I didn't have time to tested it. I will test it in few days. I have to prepare server for that because I change my enviroment. Thank you for your time and help with my controller.