john30/ebusd #280: Problem with recognise Vaillant boiler

tedxxx2, 2019-04-25:
Hi All,
I have a problem with reading data from my Vaillant boiler. I receive following readout from ebusd log:

```
2019-04-18 20:03:20.965 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-04-18 20:03:21.599 [bus notice] bus started with own address 31/36
2019-04-18 20:03:21.600 [bus notice] signal acquired
2019-04-18 20:03:22.224 [bus notice] new master 10, master count 2
2019-04-18 20:03:22.224 [update notice] received unknown BC cmd: 10feb5160301f00e
2019-04-18 20:03:24.290 [update notice] received unknown MS cmd: 1050b509040e010001 / 00
2019-04-18 20:03:24.540 [update notice] received unknown MS cmd: 1050b509040e200001 / 00
2019-04-18 20:03:24.799 [update notice] received unknown MS cmd: 1050b5040118 / 060000009b0105
2019-04-18 20:03:25.089 [bus notice] new master 03, master count 3
2019-04-18 20:03:25.090 [update notice] received unknown MS cmd: 1008b5100900004760ffff00ff00 / 0101
2019-04-18 20:03:26.322 [update notice] received unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
2019-04-18 20:03:26.561 [update notice] received unknown MS cmd: 1050b505021800 / 00
2019-04-18 20:03:28.303 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2019-04-18 20:03:30.238 [update notice] received unknown MS cmd: 1050b5040126 / 0700000009004101
2019-04-18 20:03:30.517 [update notice] received unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2019-04-18 20:03:30.799 [update notice] received unknown MS cmd: 1008b5110101 / 093232f00e58580000ff
2019-04-18 20:03:32.255 [main error] scan config 08: ERR: arbitration lost
2019-04-18 20:03:32.335 [update notice] received unknown MS cmd: 1008b5110102 / 05033c965062
2019-04-18 20:03:34.305 [update notice] received unknown MS cmd: 1050b509040e010001 / 00
2019-04-18 20:03:34.556 [update notice] received unknown MS cmd: 1050b509040e200001 / 00
2019-04-18 20:03:34.816 [update notice] received unknown MS cmd: 1050b5040118 / 060000009b0105
2019-04-18 20:03:35.103 [update notice] received unknown MS cmd: 1008b5100900004760ffff00ff00 / 0101
2019-04-18 20:03:36.377 [update notice] received unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
2019-04-18 20:03:36.616 [update notice] received unknown MS cmd: 1050b505021800 / 00
2019-04-18 20:03:38.296 [main error] scan config 15: ERR: arbitration lost
2019-04-18 20:03:38.350 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2019-04-18 20:03:40.400 [update notice] received unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2019-04-18 20:03:40.712 [update notice] received unknown MS cmd: 1008b5110101 / 093232f00e58580000ff
2019-04-18 20:03:42.405 [update notice] received unknown MS cmd: 1008b5040100 / 0a00574412fffffffff00e
2019-04-18 20:03:42.598 [main error] scan config 50: ERR: arbitration lost
2019-04-18 20:03:42.630 [update notice] received unknown BC cmd: 10feb505020400
2019-04-18 20:03:44.379 [update notice] received unknown MS cmd: 1050b509040e010001 / 00
2019-04-18 20:03:44.632 [update notice] received unknown MS cmd: 1050b509040e200001 / 00
2019-04-18 20:03:44.896 [update notice] received unknown MS cmd: 1050b5040118 / 060000009b0105
2019-04-18 20:03:45.183 [update notice] received unknown MS cmd: 1008b5100900004760ffff00ff00 / 0101
2019-04-18 20:03:46.410 [update notice] received unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
2019-04-18 20:03:46.653 [update notice] received unknown MS cmd: 1050b505021800 / 00
2019-04-18 20:03:48.391 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2019-04-18 20:03:48.792 [update notice] received unknown MS cmd: 1050b5040126 / 0700000009004101
2019-04-18 20:03:50.422 [update notice] received unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2019-04-18 20:03:50.707 [update notice] received unknown MS cmd: 1008b5110101 / 093232f00e58580000ff
2019-04-18 20:03:52.406 [update notice] received unknown MS cmd: 1050b505053c59014101 / 00
2019-04-18 20:03:54.450 [update notice] received unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
2019-04-18 20:03:54.696 [update notice] received unknown MS cmd: 1050b509040e010001 / 00
2019-04-18 20:03:54.945 [update notice] received unknown MS cmd: 1050b509040e200001 / 00
2019-04-18 20:03:55.132 [main error] scan config 08: ERR: arbitration lost
2019-04-18 20:03:55.203 [update notice] received unknown MS cmd: 1050b5040118 / 060000009b0105
2019-04-18 20:03:55.488 [update notice] received unknown MS cmd: 1008b5100900004760ffff00ff00 / 0101
2019-04-18 20:03:55.732 [update notice] received unknown MS cmd: 1050b505021800 / 00
2019-04-18 20:03:56.433 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2019-04-18 20:03:58.480 [update notice] received unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2019-04-18 20:03:58.763 [update notice] received unknown MS cmd: 1008b5110101 / 093232f00e58580000ff
2019-04-18 20:04:00.456 [update notice] received unknown BC cmd: 10feb516080031032018040419
2019-04-18 20:04:00.645 [main error] scan config 15: ERR: arbitration lost
2019-04-18 20:04:00.709 [update notice] received unknown MS cmd: 1008b512020064 / 00
2019-04-18 20:04:02.457 [update notice] received unknown MS cmd: 1050b5040118 / 060000009b0105
2019-04-18 20:04:04.464 [update notice] received unknown MS cmd: 1050b503020003

```
Tried to use different setting for latency like:

`EBUSD_OPTS="--scanconfig -d /dev/ttyUSB0 -p 8888 -l /var/log/ebusdUSB0.log --latency=20000 --lograwdata=Bytes"
`

Unfortunately, that’s doesn't helped. Still Vaillant boiler cannot be recognized in that configuration.

[... 111 lines cut ...]
ebusd working on Raspberry Pi 3 B WiFi
Operating system: Linux raspberrypi 4.14.98-v7+
Checked on ebusd 2.1 and 3.3 version and on independent Debian system (Live CD) on Notebook.
Unfortunately nothing changed.

**Ebus Adapter hardware:**
Tested on Ebus adapter (FHEM) - https://wiki.fhem.de/wiki/EBUS
and on newest Ebus 2.2 - https://ebus.github.io/adapter/
Unfortunately nothing changed.

**USB-TTL Hardware adapter:**
CP2102 chipset
PL 2303HX chipset
FT232RL chipset
and also on Rasperry internal UART.

Additionally I also check communication with short (grounded) cable connected directly to Boiler - to avoid any noise from other equipment.
Unfortunately that also doesn’t helped.

FYI I have Vaillant boiler **EcoCompact2 with calorMATIC 430 and VR61 mixing controller.** All ebus communication cables are connected together in boiler housing.

I’m stick in this point since few weeks :-( and have no other idea how to handle with this problem.

Best Regards
Sebastian

john30, 2019-05-01:
it seems that your adapter is unable to write to the bus. try starting ebusd with `--lograwdata=bytes` in foreground and check if any sent ">31" byte is every received back immediately in "<31". If not, your adapter is not able to send at all.

tedxxx2, 2019-05-28:
John, you are correct!! It was a problem with cable connector for write signal (TX). I used the same cable for all my attempts on Ebus Adapters. Lesson learnt :-)
Thanks for pointing into correct direction. Now everything being recognized!

Best Regards
Sebastian

closed 2019-05-28 by tedxxx2