john30/ebusd #282: Integration with Domoticz

guillaumezin, 2019-05-01:
Hello,

Would you add in https://github.com/john30/ebusd/wiki/7.-Integrations a link to https://github.com/guillaumezin/DomoticzEbusd which allows an easy integration with ebusd?

Thank you.

john30, 2019-05-04:
could you provide a screenshot or something alike so that a user gets a first impression of what this is good for?
your readme points to port 8888 primarily (especially in the firewall section) but the python script also uses the JSON port 8889. hence my question:
why do you need the JSON port at all? the same data is also retrievable via telnet port (or did I miss something here?)

guillaumezin, 2019-05-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I use JSON to parse exact message type, this way I can tell Domoticz if the message is a temperature, a selector switch, etc. It's the only way I managed to do it, I didn't find out how to do it with Telnet only.

For the firewall, I suggest to protect port 8888 because this is the telnet port and you can read AND write messages, the JSON port is a read only port, so I didn't bother protecting it.

john30, 2019-06-10:
done

closed 2019-06-10 by john30