john30/ebusd #283: Problems with Kernel 4.19.42-v7+ (openhab 1.41)

Gyrinx, 2019-05-31:
Hallo John,

I have the problem with the above Kernel Number, that the EBUSD don't get any datas.
The only problem, I recognized is the following error during compilation.

make[1]: Verzeichnis „/home/openhabian/ebusd/src/ebusd“ wird betreten
  CXX      bushandler.o
bushandler.cpp: In member function ‘void ebusd::BusHandler::formatGrabResult(bool, bool, std::ostringstream*, bool, time_t, time_t) const’:
bushandler.cpp:1561:19: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
     if (since > 0 && it.second.getLastTime() < since
         ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

john30, 2019-06-02:
this is just a warning, no error, and does not have any influence on the generated binary.
what is your interface you connect to and how is it connected? via USB, Ethernet, serial? what is the hardware you'd like to run ebusd on?

SailingTom, 2019-06-10:
Hi John,

sorry if i placed the wrong question :-(

Is there a solution ?

It seems to be an issue with the /dev/ttyebus !

Greets
Tom

john30, 2019-06-10:
could you check access rights on the /dev/ttyebus device?

SailingTom, 2019-06-10:
pi@fhem-pi:~/ttyebus $ ls -la /dev/ttyebus
crw-rw-rw- 1 root root 10, 57 Jun 10 10:46 /dev/ttyebus

after installing the ttyebus - shown here  - https://github.com/eBUS/ttyebus

Greets
Tom

john30, 2019-06-10:
ok, so it's no right issue. let's see what @Galileo53 finds out

SailingTom, 2019-06-10:
John,

OK - so will play the "Jeopardy Theme" for a while. (till Galileo53 replies :-))

Thank You

john30, 2019-06-10:
for the time being, you could downgrade the kernel to get it working again I guess

Gyrinx, 2019-06-10:
Aktual, I use the origin software driver /dev/ttyAMA0, which works with the new kernel without any issues. I only use the ebusd for logging data. Maybe there are issues with the origin driver and writing back to Vaillant.

I have a Vaillant 196/2 with a VC430.

Kind regards and thank you for your good work, John.
Dirk

Gyrinx, 2019-06-10:
Additional, I use the V2.2 Hardware Module.

SailingTom, 2019-06-10:
OK - uninstalled ttyebus and activated ttyAMA0

This is what the EBUSD says:

 ebusd -f --scanconfig -d /dev/ttyAMA0 --latency=20000 --lograwdata=byte
2019-06-10 13:36:02.784 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-06-10 13:36:03.102 [bus notice] bus started with own address 31/36
2019-06-10 13:36:03.157 [bus notice] signal acquired
2019-06-10 13:36:07.837 [bus notice] new master 10, master count 2
2019-06-10 13:36:07.918 [bus notice] new master 03, master count 3
2019-06-10 13:36:07.919 [update notice] received unknown MS cmd: 1008b5110101 / 0982ff101700780000ff
2019-06-10 13:36:07.919 [bus notice] <1008b511010189000982ff101700780000ff8600
2019-06-10 13:36:08.204 [update notice] received unknown MS cmd: 1008b5100900000078ffff010000 / 0101
2019-06-10 13:36:08.204 [bus notice] <1008b5100900000078ffff010000830001019a00
2019-06-10 13:36:13.183 [bus notice] >310807
2019-06-10 13:36:13.183 [main error] scan config 08: ERR: SYN received
2019-06-10 13:36:15.229 [bus notice] >311507
2019-06-10 13:36:15.229 [main error] scan config 15: ERR: SYN received
2019-06-10 13:36:17.954 [update notice] received unknown MS cmd: 1008b5110101 / 0982ff101700780000ff
2019-06-10 13:36:17.954 [bus notice] <1008b511010189000982ff101700780000ff8600
2019-06-10 13:36:18.235 [bus notice] <1008b5100900000078ffff010000830001019a00
2019-06-10 13:36:27.276 [bus notice] >310807
2019-06-10 13:36:27.276 [main error] scan config 08: ERR: SYN received
2019-06-10 13:36:28.042 [update notice] received unknown MS cmd: 1008b5110101 / 0982ff101700780000ff
2019-06-10 13:36:28.042 [bus notice] <1008b511010189000982ff101700780000ff8600
2019-06-10 13:36:28.322 [bus notice] <1008b5100900000078ffff010000830001019a00
2019-06-10 13:36:28.604 [bus notice] <1008b50401003d000a033036131006011910176e00
2019-06-10 13:36:28.865 [bus notice] <1008b51101028a0005033c9650788400
2019-06-10 13:36:29.094 [update notice] received unknown BC cmd: 10feb516080028361310060119
2019-06-10 13:36:29.094 [bus notice] <10feb516080028361310060119ec
2019-06-10 13:36:29.379 [update notice] received unknown MS cmd: 1008b5110100 / 0812040f001f000080
2019-06-10 13:36:29.379 [bus notice] >31<1008b511010088000812040f001f0000805000
2019-06-10 13:36:29.513 [bus notice] >311507
2019-06-10 13:36:29.513 [main error] scan config 15: ERR: SYN received
2019-06-10 13:36:29.736 [update notice] received unknown BC cmd: 10feb51603011017
2019-06-10 13:36:29.736 [bus notice] <10feb51603011017f9
2019-06-10 13:36:38.112 [update notice] received unknown MS cmd: 1008b5110101 / 0982ff101700780000ff
2019-06-10 13:36:38.113 [bus notice] <1008b511010189000982ff101700780000ff8600
2019-06-10 13:36:38.390 [bus notice] <1008b5100900000078ffff010000830001019a00
2019-06-10 13:36:41.594 [bus notice] >310807
2019-06-10 13:36:41.595 [main error] scan config 08: ERR: SYN received
2019-06-10 13:36:43.643 [bus notice] >311507
2019-06-10 13:36:43.643 [main error] scan config 15: ERR: SYN received
2019-06-10 13:36:48.158 [bus notice] <1008b511010189000982ff101700780000ff8600
2019-06-10 13:36:48.436 [bus notice] <1008b5100900000078ffff010000830001019a00

SailingTom, 2019-06-10:
> Aktual, I use the origin software driver /dev/ttyAMA0, which works with the new kernel without any issues. I only use the ebusd for logging data. Maybe there are issues with the origin driver and [...]

Hi Dirk,

can you post your EBUSD_OPTS ?

Thanks
Tom

Gyrinx, 2019-06-10:
EBUSD_OPTS="--scanconfig -device=/dev/ttyAMA0 --latency=20000 --httpport=8080 /var/log/ebusd.log  --loglevel=info"

Maybe you have problems with latency?
Do you have deactivate Bluetooth?

sudo echo "dtoverlay=pi3-miniuart-bt" >> /boot/config.txt

Good luck,
Dirk

Gyrinx, 2019-06-10:
Do you have installed the ebus config files? 
You output looks like mine, when I forget to install the config file ;)

Dirk

SailingTom, 2019-06-10:
> EBUSD_OPTS="--scanconfig -device=/dev/ttyAMA0 --latency=20000 --httpport=8080 /var/log/ebusd.log --loglevel=info"
> 
> Maybe you have problems with latency?
> Do you have deactivate Bluetooth?
> 
> sudo echo "dtoverlay=pi3-miniuart-bt" >> /boot/config.txt
> 
> Good luck,
> Dirk

Hi Dirk,

here is my config.txt

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
program_usb_boot_mode=1
enable_uart=1
#
dtoverlay=pi3-miniuart-bt
enable_uart=1
start_x=1

SailingTom, 2019-06-10:
> Do you have installed the ebus config files?
> You output looks like mine, when I forget to install the config file ;)
> 
> Dirk

Hi again,

i thought the files a sacanned from http://ebusd.eu/config/

But i will install the config files in etc/ebusd / version 2.1 ?

Regards
Tom

john30, 2019-06-13:
yes, with a recent ebusd configs can be loaded from the webservice unless you explicitly set the config file location in the ebusd start params

Galileo53, 2019-06-17:
The interrupt number for the "uart-pl011" was changed from 87 to 81 at kernel version v4.19.42-v7+.
A new version 1.6 of the ttyebus is available, that reflects this change.
Please fetch this version and rebuild the driver.
Regards,
Eduard

SailingTom, 2019-06-17:
Moin,

 

steh ich auf der Leitung ?

 

 

 

Laut Code in der ttyebusm.c zeigt der aber bei der 4.19.42 auf den IRQ 87 und
bei allen anderen auf  IRQ 81.

Das muss doch genau anders herum sein. Oder ?

 

 

Gruß

 

Thomas

(#283)

 

<https://github.com/john30/ebusd/issues/283?email_source=notifications&email_
token=AMJVM3BLU6NZPHQV47FF6P3P24KGFA5CNFSM4HR253UKYY3PNVWWK3TUL52HS4DFVREXG43
VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX2A6XY#issuecomment-502533983> , or mute
the thread
KGFANCNFSM4HR253UA> . 
<https://github.com/notifications/beacon/AMJVM3GED5U7NHWYFO5TN3TP24KGFA5CNFSM
4HR253UKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX2A6XY.g
if> 

Wie Sie wissen, können über das Internet versandte E-Mails unter fremden Namen erstellt oder manipuliert werden. Aus diesem Grund sind unsere mit E-Mail verschickten Nachrichten grundsätzlich keine rechtsverbindlichen Willenserklärungen.

Diese E-Mail wurde am 17/6/2019 von Thomas.Darius@darius-net.de an recipient gesendet.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Galileo53, 2019-06-17:
Hallo Thomas,
#if LINUX_VERSION_CODE < KERNEL_VERSION(4,19,42)
#define RASPI_23_UART_IRQ 87
bedeutet dass alle Versionen *KLEINER ALS* den IRQ 87 bekommen.
Die Version 4.19.42 und alle höheren bekommen also 81.
LG
Eduard

SailingTom, 2019-06-17:
Hallo Eduard,

 

danke für die Aufklärung und deine Arbeit. Ich werde das heute Abend testen.

 

Gruß

 

Thomas

(#283)

 

<https://github.com/john30/ebusd/issues/283?email_source=notifications&email_
token=AMJVM3CQ3GG4NDJ5TZXIOQ3P257QHA5CNFSM4HR253UKYY3PNVWWK3TUL52HS4DFVREXG43
VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX3AI6A#issuecomment-502662264> , or mute
the thread
7QHANCNFSM4HR253UA> .
<https://github.com/notifications/beacon/AMJVM3DPI36PQQRBJLQXJELP257QHA5CNFSM
4HR253UKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX3AI6A.g

https://www.avast.com/antivirus

SailingTom, 2019-06-17:
Update:

uninstalled EBUSD and ttyebus.

Newly make with actual Versions from Github.

Same Problem :-(

ebusctl i

version: ebusd 3.3.v3.3-19-ga3f4999
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

Any further suggestions ?

Regards 

Thomas

Galileo53, 2019-06-17:
We can debug this:
1. What kernel version do you have?
   uname -r
2. uninstall the ttyebus
   sudo service ebusd stop
   sudo make uninstall
3. re-activate ttyAMA0
   remove (or comment) the line "dtoverlay=pi3-miniuart-bt" from /boot/config.txt
   (assuming you have an raspi3)
4. reboot
5. see where the uart-pl011 is originally assigned.
cat /proc/interrupts
This is how it looks at my raspi:
[image]
You can see it is on 81.
Don't forget to re-activate the "dtoverlay=pi3-miniuart-bt" afterwards :-)
Regards,
Eduard

SailingTom, 2019-06-17:
Hi Galileo,

I have followed the step you asked.

Starting with my Raspi 3 with Kernel 4.14.79

[image]

then after upgrade to Kernel 4.19.42

[image]

after installing ttyebus

[image]

but !!!

[image]

Any Ideas left ?

Greetz
Thomas

Galileo53, 2019-06-18:
Hi Thomas,
Imho, the ttyebus should work correctly now.
But anyway, what you can do is uncomment "#define DEBUG 1" at line 57 of ttyebusm.c.
then rebuild and run and have a look at /var/log/kern.log and search for "ttyebus" entries.
Maybe you can find some ttyebus error messages.
Regards, Eduard

SailingTom, 2019-06-18:
Good Morning,

dont know what happens now. After a few REBOOTS the ttyebus now shows:

pi@fhem-pi:~ $ ebusctl i
version: ebusd 3.3.v3.3
signal: acquired
symbol rate: 118
max symbol rate: 118
min arbitration micros: 3
max arbitration micros: 23
min symbol latency: 4
max symbol latency: 21
reconnects: 0
masters: 3
messages: 231
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010007508.inc" ([PROD='0010007692']), "vaillant/08.bai.csv"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

How can I check if EVERYTHING is OK now ?

Greetz

Thomas

andig, 2019-11-04:
Looks like this has been fixed in ttyebus via https://github.com/eBUS/ttyebus/commit/49f4cf6de04e4e77bfb7904e91c842dc70074476 and can be closed?

john30, 2019-11-05:
@Galileo53 can you check if this is closable?

MarkSchmitt, 2019-12-27:
@john30 seems to be ttyebus-driver issue.

after updating to kernel 4.19.66-v7+ on raspbian stretch ebusd (updated to 3.4.v3.3-51-g57eae05 using "deb https://www.ebusd.eu/repos/apt/default/stretch stretch main" as package source ebusd would not log any bus messages. took me the better part of the morning, but it looks like updating the ttyebus driver fixed the issue. doesn't seem to be related to ebusd at all but a driver issue with the 4.19-line.

how to fix:
in the directory where you checked out https://github.com/ebus/ttyebus.git
`make uninstall (you need to stop ebusd or anything accessing /dev/ttyebus, f.e. systemctl stop ebusd)`
`make clean install`
`modprobe ttyebus`

ebusd should be able to use the driver again.

john30, 2019-12-30:
ok thanks

closed 2019-12-30 by john30