john30/ebusd #284: Not responding after starting ebusd scanconfig

diamond95, 2019-05-31:
Hi John, 

I have WRSol2.1 and after installation on debian, I'm trying to run:
cd ebusd
sudo ebusd -f -d /dev/ttyUSB0 --scanconfig
...
[main notice] ebusd 3.1 ... started with auto scan

After that message, nothing shows up, even after 30 minutes.
Am I missing something? 
I also tried to check ebusd status, and get this: https://www.photobox.co.uk/my/photo/full?photo_id=501900080025

Best regards

john30, 2019-06-02:
the status is ok. is your adapter connected to /dev/ttyUSB0?

diamond95, 2019-06-02:
Yes it is..
https://imgur.com/a/ycRiOip

john30, 2019-06-02:
this adapter is unknown to me and probably does not forward the raw data to the serial device

diamond95, 2019-06-03:
Which one do you suggest? 
On photo you can see I have some sort of converter called Ebus, from device to pc, but didnt work out.. :/

john30, 2019-06-10:
see the wiki https://github.com/john30/ebusd/wiki/6.-Hardware

closed 2019-06-10 by john30