john30/ebusd #286: Emulate outdoor temperature sensor via ebusd

davefx, 2019-07-22:
Can we emulate an EBUS outdoor temperature sensor device via ebusd?

I don't have this kind of hardware, and having access to the external temperature it would be really nice if I can emulate it and provide outdoor temperature to the boiler.

Dinth, 2019-07-22:
I would also be interested in this. Ive got Vaillant VRC 470f but since my boiler is in the loft, installing external sensors (which comes with VRC470f) is very tricky.

john30, 2019-08-11:
in principle this should be possible. But usually the message depends highly on the environment. you could start with making ebusd regularly send the eBUS defined broadcast datetime message that also includes the outside temperature. but it should not interfere with another component already sending the same message, so it might really be tricky

Dinth, 2019-08-11:
What is the correct datetime format with an outside temperature included? I already have an automation which automatically syncs the time with an NTP server.

john30, 2019-08-15:
that of course depends on the message defintion.
for Vaillant broadcast it is this: [vaillant/broadcast.csv:4](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/broadcast.csv)
and you'd probably also have to send the standard broadcast: [broadcast.csv:5](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/broadcast.csv)

salvq, 2020-10-25:
@davefx, @Dinth  Guys, have you successfully emulated outdoor temperature, i.e. sending MQTT msg to topic in certain periodicity ? If so, please share.

**This is broadcast I am seeing in ebus**
```
test_ebus/broadcast/outsidetemp {
     "temp2": {"value": 11.562}}
```

andig, 2020-10-26:
I don't think this approach is gonna work. Reasoning (deducted from my environment): the sensor is attached to the VRC700. The sensor is not its own bus device, although the readings are echoed by the VRC. The VRC controls the boiler via FlowTempDesired based on the measured temperature. I'm assuming that- since the sensor it attached to the VRC- it will only take measurements from attached sensors into account.

salvq, 2020-10-26:
well, it makes sense what you just said....however:

- Question is, why I am seeing broadcast every minute on ebus ?
- What is the purpose of this broadcast ?
- Is there any reference how the external sensor works ?

**Every minute broadcast**
```
2020-10-26 23:37:01.162 test_ebus/broadcast/outsidetemp {
2020-10-26 23:37:01.177      "temp2": {"value": 11.125}}
2020-10-26 23:38:11.354 test_ebus/broadcast/outsidetemp {
2020-10-26 23:38:11.369      "temp2": {"value": 11.125}}
2020-10-26 23:39:10.583 test_ebus/broadcast/outsidetemp {
2020-10-26 23:39:10.598      "temp2": {"value": 11.125}}
2020-10-26 23:40:10.888 test_ebus/broadcast/outsidetemp {
2020-10-26 23:40:10.903      "temp2": {"value": 11.125}}
2020-10-26 23:41:11.050 test_ebus/broadcast/outsidetemp {
2020-10-26 23:41:11.065      "temp2": {"value": 11.125}}
2020-10-26 23:42:11.210 test_ebus/broadcast/outsidetemp {
2020-10-26 23:42:11.225      "temp2": {"value": 11.125}}
```

andig, 2020-10-27:
There are no references for anything. And why shouldn‘t the vrc broadbast rhe temp? There are even broadcasts for time and date ;)

closed 2021-10-06 by john30

boelle, 2023-07-13:
Q: how could i make a hardware sensor diy?

boelle, 2023-07-13:
ahh yes.... VRC700, the tech also said that it could be remote mounted.... i could remember wrong

but i would more like a small discrete sensor placed central and not a big ugly display

Dinth, 2023-07-13:
what do you mean by remotely mounted? vrc700 should be installed in a room which s being heated. You can add extra sensors to different room as well (vr91)

boelle, 2023-07-13:
well its installed in the boiler, and it was that when i got the house... So the installer F'ed that up

Dinth, 2023-07-13:
> well its installed in the boiler, and it was that when i got the house... So the installer F'ed that up

hmmm i didnt know that vrc700 can be installed in the boiler, i thought its only an option for vrc700f basestation.
You can just take it out, hang it on aa wall and connect via ebus, or buy an extra vr91

boelle, 2023-07-13:
hehe.... the problem is the better half would hate "a big ugly display that we never use"

i guess vaillant do not make a true sensor only, ie something very small  that could be in a small 50x50mm white box

that is why i initial asked if i could make something myself, i have searched for ebus temperature sensors but came up empty