john30/ebusd #288: Fix compilation on the FreeBSD

samm-git, 2019-08-04:
Hi,

I was trying to compile ebusd on freebsd and found that some includes are missing.

Here is a patch:

```
diff --git a/src/lib/ebus/device.h b/src/lib/ebus/device.h
index 7c555f8..115fc36 100755
--- a/src/lib/ebus/device.h
+++ b/src/lib/ebus/device.h
@@ -25,6 +25,7 @@
 #include <netdb.h>
 #include <iostream>
 #include <fstream>
+#include <netinet/in.h>
 #include "lib/ebus/result.h"
 #include "lib/ebus/symbol.h"

diff --git a/src/lib/utils/tcpsocket.h b/src/lib/utils/tcpsocket.h
index 4d0dcb1..fd82b22 100755
--- a/src/lib/utils/tcpsocket.h
+++ b/src/lib/utils/tcpsocket.h
@@ -21,6 +21,7 @@

 #include <unistd.h>
 #include <sys/socket.h>
+#include <netinet/in.h>
 #include <sys/time.h>
 #include <stdint.h>
 #include <string>
```

samm-git, 2019-08-05:
P.S. i also found that serial port on freebsd is not working, will do another patch soon

samm-git, 2019-08-05:
Patch added to the https://github.com/john30/ebusd/pull/290

john30, 2019-08-11:
closed with PR #290

closed 2019-08-11 by john30